gmag11/EnigmaIOT #1: Feature Request : include IR protocol with ESP-Now protocol

happytm, 2019-08-19:
There are very few implementation of ESP-Now protocol on internet so I am glad I found your repository here. Is it possible to implement IR communication between sensor & gateway in your code? It will be useful if sensor and gateway are indoor in same room then we can use IR protocol and that way it will help to add more sensors to gateway beyond the limit of 6 ESP-Now sensors to each gateway.

Thanks

gmag11, 2019-08-28:
There is no such limit of 6 nodes. That is because EnigmaIOT does not use esp-now encryption but its own crypto layer. Indeed my plan is to install about 20 sensors.

In theory it would be no limitation apart from memory on gateway. It needs 60 bytes for each node... and it has about 30 kB still free!!!

I've not made current tests yet. But time on is about 300 ms with an expected current of about 100mA. While in sleep mode, ESP8266 specificates 12 uA. I guess I will not be able to achieve that but I home to get about 100 uA.

My will is to have 2xAA batteries node working for 1 year.

gmag11, 2019-08-28:
About IR. In theory it could be done, but I do not see the use case. Maybe I have not understood what you mean.

If you want to communicate with an IR remote, then you need to program an EnigmaIOT node that understands IR commands and send them over esp-now.

If you want to design a node that communicates with gateway using IR then you have to develop an implementation of [Comms_hal.h](https://github.com/gmag11/EnigmaIOT/blob/master/src/Comms_hal.h)

happytm, 2019-08-30:
Wow. that is really fast. 

For IR protocol I was thinking to have door and window sensors on Attiny85 which is powered up by normally open hall sensor switch when door opens. Hall sensor will  close the switch  every time door or window opens  and wake Attiny85 up. It will send custom IR code to EnigmaIOT gateway which will have IR receiver circuit connected to ESP8266 and then Attiny85 will go to deep sleep. It will wake up from deep sleep every 5 minutes & send IR code as long is door is open. When door is closed it will power down with help of hall switch.So sender side you do not need any extra code but gateway should be able to receive and decode IR message from door sensor and send the message to MQTTbridge via Serial as you already have implemented for ESP-now protocol.Normally it will be same IR code from Attiny85 saying Door or window is open. If there is no message after 5 minutes gateway will assume the door or window is closed and gateway will pass message to MQTT bridge saying door is closed.

For Attiny85 I want to use following IR library:

https://gist.github.com/dquadros/2df488663eb1f59994d6d142f61ed3af

Thanks

happytm, 2019-08-30:
I looked into Comms_hal.h file but I could not understand how to implement IR sender code as I am not experienced programmer.

Thanks

happytm, 2019-09-03:
I am trying to use example of sensor but so far I am unsuccessful to compile with arduino.It is complaining about EnigmaIOTconfig.h file not found.

Can you please specify where to put all files & depenancies to compile the example code?

Thanks

happytm, 2019-09-03:
I found problem of missing library called <ESPAsyncTCP.h>

Thanks

happytm, 2019-09-03:
After many hours of moving files around I am stuck here. There must be easy way to compile this for esp8266 under arduino 2.5.0. I removed those 2 files as instructed. Please help.

C:\Users\WinCycle\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp: In constructor 'AsyncWebSocketResponse::AsyncWebSocketResponse(const String&, AsyncWebSocket*)':

C:\Users\WinCycle\Documents\Arduino\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp:1248:31: error: 'SHA1' was not declared in this scope

   sha1(key + WS_STR_UUID, hash);

Thanks

gmag11, 2019-09-24:
I guess you deleted the wrong sha1.h file. I recommend you to reinstall esp8266 core using Arduino Boards Manager.

I have good news. I've forked Crypto Library so that there is no need to delete/edit any file from it. I've modified Readme file to point to current repository https://github.com/gmag11/CryptoArduino.

happytm, 2019-09-24:
Thank you. I was looking for simple data transfer protocol between master & slave ESP devices.My goal was to send sensor data from battery powered node to main controller within 200 milliseconds and I was hopeful I could do it using ESPNow but I have now found more simpler solution to achieve my goal without using ESPNow. I am able to send data in roughly 220 milliseconds using adhoc network.

Thanks again.

gmag11, 2019-09-25:
Great. I close the issue then. :)

closed 2019-09-25 by gmag11