gmag11/EnigmaIOT #5: Problem with nodes of more than one sensor.

dgcasana, 2020-05-01:
Before with version 0.5.1 there was no problem with a bme280 connected to the node.
But with version 0.8.1 / 0.8.2
the mqtt is not sent complete.
this is the DBG output:
[609330] 24392 free (processRxData:162) Published data message from A0:20:A6:08:CD:88: [{"channel":38,"type":103,"name":"temperature","value":13.6},{"channel":38,"type":104,"name":"humidity","value":99},{"channel":38,"type":142,"name":"switch","value":0},{"channel":38,"type":115,"name":"pressure","value":0.9},{"channel":38,"type":142,"name":"switch","value":0},{"channel":39,"type":2,"name":"analog_input","value":246}]
And mqtt
I [609415] 24760 free (publishMQTT:411) Publish MQTT. Esp-Now/A0:20:A6:08:CD:88/data : [{"channel":38,"type":103,"name":"temperature","value":13.6},{"channel":38,"ty

I have tried to change the size of the mqtt buffer but it gets worse, I use esp8266.
Thank you.

P.D.
I use the channel number as Domoticz idx, and the switch type as dummy data for domoticz.

gmag11, 2020-05-02:
I will check thank you

gmag11, 2020-05-03:
I was using a 8 bit variable to store MQTT payload length. So, for length bigger than 255 it was overflowed.

I've fixed payload data and control data methods to use `size_t` type.

Test code on https://github.com/gmag11/EnigmaIOT/commit/c2d1ff45d3903c7f142179811b8b13c8453eb20c

Notice that you need to update `GwOutput_mqtt.cpp`and `GwOutput_mqtt.h` in your sketch from files in library example.

dgcasana, 2020-05-04:
Tested, works fine. Thank you very much. I have changed all my mqtt sensors for your enigmaiot, with your library it is very easy to create new sensors, currently I use 5 nodes.

gmag11, 2020-05-05:
Good to know! I will do a fast release to include this fix

gmag11, 2020-05-05:
Fixed on release [0.8.3](https://github.com/gmag11/EnigmaIOT/tree/0.8.3)

closed 2020-05-05 by gmag11