gmag11/EnigmaIOT #6: Infinite loop

dgcasana, 2020-05-11:
I have been testing v0.8.3 and I only receive the status and the rssi but not the message.
I have compiled the gateway with debug_info and this is what I get:
[I][EnigmaIOTGatewayMQTT.ino:174] processRxData(): 7725 Heap: 237524 Published MQTT from A0:20:A6:08:CD:88: {"per":0.000000e+00,"lostmessages":0,"totalmessages":1,"packetshour":11320.75}
[I][GwOutput_mqtt.cpp:411] publishMQTT(): 7764 Heap: 237772 Publish MQTT. Esp-New/A0:20:A6:08:CD:88/data : [{"channel":38,"type":103,"name":"temperature","value":27.4},{"channel":38,"type":104,"name":"humidity","value":69},{"channel":38,"type":142,"name":"switch","value":0},{"channel":38,"type":115,"name":"pressure","value":0.9},{"channel":38,"type":142,"name":"switch","value":0},{"channel":39,"type":2,"name":"analog_input","value":42}]
[I][EnigmaIOTGateway.cpp:836] manageMessage(): 7780 Heap: 238048 Data OK
[I][GwOutput_mqtt.cpp:411] publishMQTT(): 7819 Heap: 238412 Publish MQTT. Esp-New/A0:20:A6:08:CD:88/data : [{"channel":38,"type":103,"name":"temperature","value":27.4},{"channel":38,"type":104,"name":"humidity","value":69},{"channel":38,"type":142,"name":"switch","value":0},{"channel":38,"type":115,"name":"pressure","value":0.9},{"channel":38,"type":142,"name":"switch","value":0},{"channel":39,"type":2,"name":"analog_input","value":42}]
[I][GwOutput_mqtt.cpp:411] publishMQTT(): 7863 Heap: 240256 Publish MQTT. Esp-New/A0:20:A6:08:CD:88/data : [{"channel":38,"type":103,"name":"temperature","value":27.4},{"channel":38,"type":104,"name":"humidity","value":69},{"channel":38,"type":142,"name":"switch","value":0},{"channel":38,"type":115,"name":"pressure","value":0.9},{"channel":38,"type":142,"name":"switch","value":0},{"channel":39,"type":2,"name":"analog_input","value":42}]
[I][GwOutput_mqtt.cpp:411] publishMQTT(): 7902 Heap: 240256 Publish MQTT. Esp-New/A0:20:A6:08:CD:88/data : [{"channel":38,"type":103,"name":"temperature","value":27.4},{"channel":38,"type":104,"name":"humidity","value":69},{"channel":38,"type":142,"name":"switch","value":0},{"channel":38,"type":115,"name":"pressure","value":0.9},{"channel":38,"type":142,"name":"switch","value":0},{"channel":39,"type":2,"name":"analog_input","value":42}]
[I][GwOutput_mqtt.cpp:411] publishMQTT(): 7940 Heap: 240256 Publish MQTT. Esp-New/A0:20:A6:08:CD:88/data : [{"channel":38,"type":103,"name":"temperature","value":27.4},{"channel":38,"type":104,"name":"humidity","value":69},{"channel":38,"type":142,"name":"switch","value":0},{"channel":38,"type":115,"name":"pressure","value":0.9},{"channel":38,"type":142,"name":"switch","value":0},{"channel":39,"type":2,"name":"analog_input","value":42}]

...infinite

I have tried with Wemos D1 mini
and with LOLIN32 (without mqtt_secure)
both with the same result.

dgcasana, 2020-05-12:
I have recompiled version 0.8.3 with dEBUG_LEVEL DBG
In Esp32, this is the output:

[D][EnigmaIOTGateway.cpp:703] addInputMsgQueue(): 91226 Heap: 237536 Message 0x01 added from B4:E6:2D:44:A0:E9. Size: 1
[D][EnigmaIOTGateway.h:186] front(): 91237 Heap: 237536 Read element. ReadIdx: 0. WriteIdx: 1. Size: 1

[D][EnigmaIOTGateway.cpp:727] getInputMsgQueue(): 91246 Heap: 239244 EnigmaIOT message got from queue. Size: 1
[D][EnigmaIOTGateway.h:168] pop(): 91256 Heap: 239244 Remove element. Buffer was not empty

[D][EnigmaIOTGateway.h:169] pop(): 91264 Heap: 239244 Before -- > ReadIdx: 0. WriteIdx: 1. Size: 1

[D][EnigmaIOTGateway.h:177] pop(): 91273 Heap: 237536 After -- > ReadIdx: 1. WriteIdx: 1. Size: 0

[D][EnigmaIOTGateway.cpp:751] popInputMsgQueue(): 91281 Heap: 239244 EnigmaIOT message pop. Size 0
[D][EnigmaIOTGateway.cpp:835] handle(): 91290 Heap: 239244 EnigmaIOT input message from queue. MsgType: 0x01
[I][EnigmaIOTGateway.cpp:845] manageMessage(): 91300 Heap: 239244 Reveived message. Origin MAC: B4:E6:2D:44:A0:E9
[I][EnigmaIOTGateway.cpp:919] manageMessage(): 91310 Heap: 239244  <------- ENCRYPTED DATA
[D][GwOutput_mqtt.cpp:458] addMQTTqueue(): 91318 Heap: 238364 1 MQTT messages queued Len:175 Esp-New/B4:E6:2D:44:A0:E9/data [{"channel":8,"type":103,"name":"temperature","value":85},{"channel":9,"type":103,"name":"temperature","value":93.1},{"channel":10,"type":103,"name":"temperature","value":85}]
[I][GwOutput_mqtt.cpp:506] outputDataSend(): 91344 Heap: 238364 MQTT queued Esp-New/B4:E6:2D:44:A0:E9/data. Length 175
[I][EnigmaIOTGatewayMQTT.ino:163] processRxData(): 91354 Heap: 238444 Published data message from B4:E6:2D:44:A0:E9, length 175: [{"channel":8,"type":103,"name":"temperature","value":85},{"channel":9,"type":103,"name":"temperature","value":93.1},{"channel":10,"type":103,"name":"temperature","value":85}]
[D][GwOutput_mqtt.cpp:458] addMQTTqueue(): 91381 Heap: 237928 2 MQTT messages queued Len:75 Esp-New/B4:E6:2D:44:A0:E9/status {"per":0.000000e+00,"lostmessages":0,"totalmessages":1,"packetshour":55.12}
[I][GwOutput_mqtt.cpp:506] outputDataSend(): 91398 Heap: 237928 MQTT queued Esp-New/B4:E6:2D:44:A0:E9/status. Length 75
[I][EnigmaIOTGatewayMQTT.ino:175] processRxData(): 91409 Heap: 238004 Published MQTT from B4:E6:2D:44:A0:E9: {"per":0.000000e+00,"lostmessages":0,"totalmessages":1,"packetshour":55.12}
[I][EnigmaIOTGateway.cpp:932] manageMessage(): 91425 Heap: 238528 Data OK
[D][GwOutput_mqtt.cpp:465] getMQTTqueue(): 91432 Heap: 238528 MQTT message got from queue
[I][GwOutput_mqtt.cpp:411] publishMQTT(): 91439 Heap: 238528 Publish MQTT. Esp-New/B4:E6:2D:44:A0:E9/data : [{"channel":8,"type":103,"name":"temperature","value":85},{"channel":9,"type":103,"name":"temperature","value":93.1},{"channel":10,"type":103,"name":"temperature","value":85}]
[D][EnigmaIOTGateway.cpp:781] handle(): 91464 Heap: 238528 EnigmaIOTGatewayClass::flashrx
[D][GwOutput_mqtt.cpp:465] getMQTTqueue(): 91472 Heap: 238528 MQTT message got from queue
[I][GwOutput_mqtt.cpp:411] publishMQTT(): 91480 Heap: 238528 Publish MQTT. Esp-New/B4:E6:2D:44:A0:E9/data : [{"channel":8,"type":103,"name":"temperature","value":85},{"channel":9,"type":103,"name":"temperature","value":93.1},{"channel":10,"type":103,"name":"temperature","value":85}]
[D][GwOutput_mqtt.cpp:465] getMQTTqueue(): 91505 Heap: 238528 MQTT message got from queue

In mosquitto I receive nothing, only this:

pi@RaspiZero:~ $ mosquitto_sub -t Esp-New/#
1
1
{"reason":4}
{"rssi":-67,"channel":6}
1
{"reason":4}
{"rssi":-76,"channel":6}

gmag11, 2020-05-12:
Hello, looking at log data I cannot conclude any error. Indeed there is no warning or error message.

Output from `mosquitto_sub` command does not correspond to that log because `{reason:4}` indicates node key revocation. I recommend to add `-v` option to see mqtt topic.

dgcasana, 2020-05-13:
This is the output of LOLIN32 with enigmaiot0.8.3 Gateway:

[I][GwOutput_mqtt.cpp:231] reconnect(): 4376 Heap: 242224 Attempting MQTT connection...
[D][GwOutput_mqtt.cpp:237] reconnect(): 4384 Heap: 242224 Clock set.
[D][GwOutput_mqtt.cpp:239] reconnect(): 4390 Heap: 242224 Connect to MQTT server: user Esp32, pass , topic Esp-New/gateway/status
[W][GwOutput_mqtt.cpp:242] reconnect(): 4489 Heap: 239812 MQTT connected
[I][GwOutput_mqtt.cpp:411] publishMQTT(): 4489 Heap: 239812 Publish MQTT. Esp-New/gateway/status : 1
[D][EnigmaIOTGateway.h:142] push(): 94157 Heap: 237916 Add element. Buffer was not full

[D][EnigmaIOTGateway.h:143] push(): 94157 Heap: 237916 Before -- > ReadIdx: 0. WriteIdx: 0. Size: 0

[D][EnigmaIOTGateway.h:158] push(): 94163 Heap: 237916 After -- > ReadIdx: 0. WriteIdx: 1. Size: 1

[D][EnigmaIOTGateway.cpp:703] addInputMsgQueue(): 94172 Heap: 237916 Message 0xFF added from 2C:F4:32:4A:08:00. Size: 1
[D][EnigmaIOTGateway.h:186] front(): 94182 Heap: 237916 Read element. ReadIdx: 0. WriteIdx: 1. Size: 1

[D][EnigmaIOTGateway.cpp:727] getInputMsgQueue(): 94191 Heap: 239624 EnigmaIOT message got from queue. Size: 1
[D][EnigmaIOTGateway.h:168] pop(): 94201 Heap: 239624 Remove element. Buffer was not empty

[D][EnigmaIOTGateway.h:169] pop(): 94209 Heap: 239624 Before -- > ReadIdx: 0. WriteIdx: 1. Size: 1

[D][EnigmaIOTGateway.h:177] pop(): 94218 Heap: 239624 After -- > ReadIdx: 1. WriteIdx: 1. Size: 0

[D][EnigmaIOTGateway.cpp:751] popInputMsgQueue(): 94227 Heap: 239624 EnigmaIOT message pop. Size 0
[D][EnigmaIOTGateway.cpp:835] handle(): 94236 Heap: 239624 EnigmaIOT input message from queue. MsgType: 0xFF
[I][EnigmaIOTGateway.cpp:845] manageMessage(): 94245 Heap: 239624 Reveived message. Origin MAC: 2C:F4:32:4A:08:00
[I][EnigmaIOTGateway.cpp:863] manageMessage(): 94255 Heap: 239624  <------- CLIENT HELLO
[D][EnigmaIOTGateway.cpp:1374] processClientHello(): 94323 Heap: 239624 Node key: 78 B8 BB 26 73 6E EA 13 B1 D6 9F 74 62 E9 63 5D B7 43 9F 31 2F E9 33 01 A1 31 C5 B6 A8 6E 5B 4E 
[I][EnigmaIOTGateway.cpp:1565] serverHello(): 94328 Heap: 239624  -------> SERVER_HELLO
[D][espnow_hal.cpp:103] send(): ESP-NOW message to 2C:F4:32:4A:08:00
[D][espnow_hal.cpp:114] send(): Peer added on channel 6. Result 0
[D][espnow_hal.cpp:125] send(): esp now send result = 0
[D][espnow_hal.cpp:128] send(): Peer deleted. Result 0
[I][EnigmaIOTGateway.cpp:1567] serverHello(): 94358 Heap: 239624 Server Hello message sent to 2C:F4:32:4A:08:00
[I][EnigmaIOTGateway.cpp:868] manageMessage(): 94367 Heap: 239624 Server Hello sent
[D][GwOutput_mqtt.cpp:458] addMQTTqueue(): 94375 Heap: 239452 1 MQTT messages queued Len:0 Esp-New/2C:F4:32:4A:08:00/hello 
[I][GwOutput_mqtt.cpp:601] newNodeSend(): 94386 Heap: 239452 Published MQTT Esp-New/2C:F4:32:4A:08:00/hello
[D][EnigmaIOTGatewayMQTT.ino:207] newNodeConnected(): 94395 Heap: 239528 New node 2C:F4:32:4A:08:00 message sent
	MAC Address: 2C:F4:32:4A:08:00
	Last counter: 0
	Key valid from: 34 ms ago
	Key: Valid
	Status: Registered. Wait for messages

[D][GwOutput_mqtt.cpp:465] getMQTTqueue(): 94417 Heap: 239528 MQTT message got from queue
[I][GwOutput_mqtt.cpp:411] publishMQTT(): 94424 Heap: 239528 Publish MQTT. Esp-New/2C:F4:32:4A:08:00/hello : 
[D][GwOutput_mqtt.cpp:403] loop(): 94435 Heap: 237572 MQTT published. Esp-New/2C:F4:32:4A:08:00/hello 
[D][GwOutput_mqtt.cpp:486] popMQTTqueue(): 94443 Heap: 238036 MQTT message pop. Size 0
[D][EnigmaIOTGateway.cpp:781] handle(): 94451 Heap: 238036 EnigmaIOTGatewayClass::flashrx
[D][EnigmaIOTGateway.h:142] push(): 95050 Heap: 237916 Add element. Buffer was not full

[D][EnigmaIOTGateway.h:143] push(): 95050 Heap: 237916 Before -- > ReadIdx: 1. WriteIdx: 1. Size: 0

[D][EnigmaIOTGateway.h:158] push(): 95056 Heap: 237916 After -- > ReadIdx: 1. WriteIdx: 2. Size: 1

[D][EnigmaIOTGateway.cpp:703] addInputMsgQueue(): 95065 Heap: 237916 Message 0x01 added from 2C:F4:32:4A:08:00. Size: 1
[D][EnigmaIOTGateway.h:186] front(): 95075 Heap: 237916 Read element. ReadIdx: 1. WriteIdx: 2. Size: 1

[D][EnigmaIOTGateway.cpp:727] getInputMsgQueue(): 95085 Heap: 239624 EnigmaIOT message got from queue. Size: 1
[D][EnigmaIOTGateway.h:168] pop(): 95094 Heap: 239624 Remove element. Buffer was not empty

[D][EnigmaIOTGateway.h:169] pop(): 95102 Heap: 239624 Before -- > ReadIdx: 1. WriteIdx: 2. Size: 1

[D][EnigmaIOTGateway.h:177] pop(): 95111 Heap: 239624 After -- > ReadIdx: 2. WriteIdx: 2. Size: 0

[D][EnigmaIOTGateway.cpp:751] popInputMsgQueue(): 95120 Heap: 239624 EnigmaIOT message pop. Size 0
[D][EnigmaIOTGateway.cpp:835] handle(): 95129 Heap: 239624 EnigmaIOT input message from queue. MsgType: 0x01
[I][EnigmaIOTGateway.cpp:845] manageMessage(): 95138 Heap: 239624 Reveived message. Origin MAC: 2C:F4:32:4A:08:00
[I][EnigmaIOTGateway.cpp:919] manageMessage(): 95148 Heap: 239624  <------- ENCRYPTED DATA
[D][GwOutput_mqtt.cpp:458] addMQTTqueue(): 95157 Heap: 238640 1 MQTT messages queued Len:280 Esp-New/2C:F4:32:4A:08:00/data [{"channel":0,"type":2,"name":"analog_in","value":3},{"channel":1,"type":103,"name":"temperature","value":20.3},{"channel":2,"type":0,"name":"digital_in","value":123},{"channel":3,"type":115,"name":"pressure","value":1007.2},{"channel":4,"type":117,"name":"current","value":2.43}]
[... 52 lines cut ...]
load:0x40080400,len:6352
entry 0x400806b8

[D][GwOutput_mqtt.cpp:114] loadConfig(): 133 Heap: 299424 Opening /mqtt.json file
[D][GwOutput_mqtt.cpp:118] loadConfig(): 133 Heap: 299168 /mqtt.json opened. 83 bytes
[... cut ...]
Esp-New/gateway/status 0
Esp-New/gateway/status 1
Esp-New/gateway/status 0
Esp-New/gateway/status 1
Esp-New/2C:F4:32:4A:08:00/hello (null)
Esp-New/gateway/status 1
** Next reboot of Node
Esp-New/2C:F4:32:4A:08:00/bye {"reason":4}
Esp-New/2C:F4:32:4A:08:00/hello (null)
Esp-New/2C:F4:32:4A:08:00/result/rssi {"rssi":-16,"channel":6}
Esp-New/gateway/status 1
Esp-New/2C:F4:32:4A:08:00/bye {"reason":4}
Esp-New/2C:F4:32:4A:08:00/hello (null)
Esp-New/2C:F4:32:4A:08:00/result/rssi {"rssi":-34,"channel":6}

gmag11, 2020-05-14:
I would say that you are sending messages too quick. As Debug is activated Serial buffer is overflowed and that's why esp32 crashes before sending first data. Try to slow down node messages and reduce debug level.

dgcasana, 2020-05-14:
Esp32 Gateway debug_level WARN:

*WM: Connecting as wifi client...
*WM: Try to connect with saved credentials
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.31.229
[W][GwOutput_mqtt.cpp:242] reconnect(): 5362 Heap: 231112 MQTT connected

Node: 0
	MAC Address: 2C:F4:32:4A:08:00
	Last counter: 0
	Key valid from: 1 ms ago
	Key: Valid
	Status: Registered. Wait for messages

dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0

pi@RaspiZero:~ $ mosquitto_sub -t Esp-New/# -v
Esp-New/gateway/status 1
Esp-New/gateway/status 0
Esp-New/gateway/status 1
Esp-New/2C:F4:32:4A:08:00/bye {"reason":4}
Esp-New/2C:F4:32:4A:08:00/hello (null)
Esp-New/2C:F4:32:4A:08:00/result/rssi {"rssi":-15,"channel":6}
Esp-New/2C:F4:32:4A:08:00/status {"per":0.000000e+00,"lostmessages":0,"totalmessages":1,"packetshour":348.84}
Esp-New/2C:F4:32:4A:08:00/status {"per":0.000000e+00,"lostmessages":0,"totalmessages":2,"packetshour":242.34}
Esp-New/2C:F4:32:4A:08:00/status {"per":0.000000e+00,"lostmessages":0,"totalmessages":3,"packetshour":178.89}
Esp-New/2C:F4:32:4A:08:00/status {"per":0.000000e+00,"lostmessages":0,"totalmessages":4,"packetshour":177.18}
Esp-New/2C:F4:32:4A:08:00/status {"per":0.000000e+00,"lostmessages":0,"totalmessages":5,"packetshour":127.78}

Node don't have reset pin connected, each time I push reset button I only get status message.

I think there is someting wrong with my libraries, but I don't know what. I use fresh arduino IDE .zip with portable folder inside and only libraries you link in github..
If you can upload  .bin firmware of Gateway I will try it.

gmag11, 2020-05-20:
I've just released a new version. Please test it. Share your code if still not working. Please use code blocks to improve readability https://guides.github.com/features/mastering-markdown/

dgcasana, 2020-05-22:
This time I have used a linux live distribution on usb to install Arduino IDE from scratch.
I have installed in .zip:
* enigma IOT 0.9.0
* ESPAsyncWiFiManager
* Arduino Crypto Library
* PubSubClient
* CayennLPP
* ArduinoJSON 6
 
when I have compiled arduino it has asked me for a library **espasyntcp.h** I have used https://github.com/me-no-dev/ESPAsyncTCP
The code used is the example # EnigmaIOTGatewayMQTT # from the enigmaIOT library in esp8266 (wemos D1 mini),
I have only modified (Linux is case-sensitive):
*  _Comms_hal.h_
#if defined (ARDUINO) && ARDUINO> = 100
#include "arduino.h" -> **"Arduino.h"**
* _GwOutput_generic.h_
#if defined (ARDUINO) && ARDUINO> = 100
#include "arduino.h" -> **"Arduino.h"**

-Debug level: Info

The node is also a wemos d1 mini with the example of the library without a bridge between D0 and RST, it only restarts when I press the reset button.

**The serial output of the Gateway:**
_W [9565] 28976 free (reconnect:242) MQTT connected
I [9565] 28976 free (publishMQTT:411) Publish MQTT. Esp-New/gateway/status : 1
I [33657] 28976 free (manageMessage:851) Reveived message. Origin MAC: B4:E6:2D:45:7A:34
I [33658] 28976 free (manageMessage:869)  <------- CLIENT HELLO
I [34253] 28976 free (serverHello:1581)  -------> SERVER_HELLO
I [34254] 28760 free (serverHello:1583) Server Hello message sent to B4:E6:2D:45:7A:34
I [34256] 28760 free (manageMessage:874) Server Hello sent
I [34261] 28624 free (newNodeSend:601) Published MQTT Esp-New/B4:E6:2D:45:7A:34/hello

Node: 0
	MAC Address: B4:E6:2D:45:7A:34
	Last counter: 0
	Key valid from: 12 ms ago
	Key: Valid
	Status: Registered. Wait for messages

I [34283] 28696 free (publishMQTT:411) Publish MQTT. Esp-New/B4:E6:2D:45:7A:34/hello : 
I [34594] 28976 free (manageMessage:851) Reveived message. Origin MAC: B4:E6:2D:45:7A:34
I [34594] 28976 free (manageMessage:925)  <------- ENCRYPTED DATA
I [34597] 27944 free (processRxData:140) CayenneLPP message
I [34603] 27520 free (outputDataSend:506) MQTT queued Esp-New/B4:E6:2D:45:7A:34/data. Length 283
I [34610] 27592 free (processRxData:175) Published data message from B4:E6:2D:45:7A:34, length 283: [{"channel":0,"type":2,"name":"analog_in","value":3.02},{"channel":1,"type":103,"name":"temperature","value":20.3},{"channel":2,"type":0,"name":"digital_in","value":123},{"channel":3,"type":115,"name":"pressure","value":1007.2},{"channel":4,"type":117,"name":"current","value":2.43}], Encoding 0x81
I [34646] 27144 free (outputDataSend:506) MQTT queued Esp-New/B4:E6:2D:45:7A:34/status. Length 78
I [34654] 27216 free (processRxData:187) Published MQTT from B4:E6:2D:45:7A:34: {"per":0.000000e+00,"lostmessages":0,"totalmessages":1,"packetshour":10746.27}
I [34668] 28248 free (manageMessage:938) Data OK
I [34673] 28248 free (publishMQTT:411) Publish MQTT. Esp-New/B4:E6:2D:45:7A:34/data : [{"channel":0,"type":2,"name":"analog_in","value":3.02},{"channel":1,"type":103,"name":"temperature","value":20.3},{"channel":2,"type":0,"name":"digital_in","value":123},{"channel":3,"type":115,"name":"pressure","value":1007.2},{"channel":4,"type":117,"name":"current","value":2.43}]
I [34705] 28248 free (publishMQTT:411) Publish MQTT. Esp-New/B4:E6:2D:45:7A:34/data : [{"channel":0,"type":2,"name":"analog_in","value":3.02},{"channel":1,"type":103,"name":"temperature","value":20.3},{"channel":2,"type":0,"name":"digital_in","value":123},{"channel":3,"type":115,"name":"pressure","value":1007.2},{"channel":4,"type":117,"name":"current","value":2.43}]
I [34736] 28248 free (publishMQTT:411) Publish MQTT. Esp-New/B4:E6:2D:45:7A:34/data : [{"channel":0,"type":2,"name":"analog_in","value":3.02},{"channel":1,"type":103,"name":"temperature","value":20.3},{"channel":2,"type":0,"name":"digital_in","value":123},{"channel":3,"type":115,"name":"pressure","value":1007.2},{"channel":4,"type":117,"name":"current","value":2.43}]__
**loop, I only press node RST button one time**
**node rst button**
_I [934273] 25672 free (manageMessage:851) Reveived message. Origin MAC: B4:E6:2D:45:7A:34
I [934281] 25672 free (manageMessage:925)  <------- ENCRYPTED DATA
I [934287] 24640 free (processRxData:140) CayenneLPP message
I [934293] 24216 free (outputDataSend:506) MQTT queued Esp-New/B4:E6:2D:45:7A:34/data. Length 283
I [934301] 24288 free (processRxData:175) Published data message from B4:E6:2D:45:7A:34, length 283: [{"channel":0,"type":2,"name":"analog_in","value":3.02},{"channel":1,"type":103,"name":"temperature","value":20.3},{"channel":2,"type":0,"name":"digital_in","value":123},{"channel":3,"type":115,"name":"pressure","value":1007.2},{"channel":4,"type":117,"name":"current","value":2.43}], Encoding 0x81
I [934336] 24072 free (outputDataSend:506) MQTT queued Esp-New/B4:E6:2D:45:7A:34/status. Length 76
I [934344] 24144 free (processRxData:187) Published MQTT from B4:E6:2D:45:7A:34: {"per":0.000000e+00,"lostmessages":0,"totalmessages":6,"packetshour":475.82}
I [934358] 25176 free (manageMessage:938) Data OK
I [934364] 25176 free (publishMQTT:411) Publish MQTT. Esp-New/B4:E6:2D:45:7A:34/status : {"per":0.000000e+00,"lostmessages":0,"totalmessages":4,"packetshour":640.57}
I [934378] 24648 free (publishMQTT:411) Publish MQTT. Esp-New/B4:E6:2D:45:7A:34/data : [{"channel":0,"type":2,"name":"analog_in","value":3.02},{"channel":1,"type":103,"name":"temperature","value":20.3},{"channel":2,"type":0,"name":"digital_in","value":123},{"channel":3,"type":115,"name":"pressure","value":1007.2},{"channel":4,"type":117,"name":"current","value":2.43}]
[... cut ...]
**node rst button**
_Esp-New/B4:E6:2D:45:7A:34/hello (null)_
**node rst button**
_Esp-New/B4:E6:2D:45:7A:34/result/rssi {"rssi":-28,"channel":6}
**node rst button**
_Esp-New/B4:E6:2D:45:7A:34/status {"per":0.000000e+00,"lostmessages":0,"totalmessages":1,"packetshour":233.24}_
**node rst button**
_Esp-New/B4:E6:2D:45:7A:34/status {"per":0.000000e+00,"lostmessages":0,"totalmessages":2,"packetshour":415.87}_

When I press the reset button of the node I only receive the status message in mqtt, never the data message.

dgcasana, 2020-05-23:
Well today I have tried node_msgpack and it **works fine !!**
Data received, and there is no infinite loop on the gateway output.
I am using the gateway that yesterday failed.

gmag11, 2020-05-26:
Still looks strange. I understand that you are using basic node example, aren't you?

gmag11, 2020-05-30:
@dgcasana  If you have more info feel free to share it here and reopen the issue

closed 2020-05-30 by gmag11

dgcasana, 2020-05-31:
yes, I use enigmaiot_node example, without changes.