gmag11/EnigmaIOT #7: Missing library "arduino.h"

huyangping, 2020-07-14:
Just downloaded the latest repository. I am having serious problems. When compiling the enigmaiot_node example in Arduino 1.8.13 IDE on a debian Linux machine I receive the following compile error:
     Arduino/libraries/EnigmaIOT-master/src/Comms_hal.h:14:21: fatal error: arduino.h: No such file or directory
 #include "arduino.h"
I have installed all of the external libraries listed on the code tab. Any useful assistance would be breatly appreciated.
Glenn G.

gmag11, 2020-07-15:
It is a case typo. Change `#include "arduino.h"` with `#include "Arduino.h"` in Comms_hal.h. I'll update it soon. Thank you for reporting

gmag11, 2020-07-15:
This should be solved in latest release https://github.com/gmag11/EnigmaIOT/releases/tag/0.9.3.1

huyangping, 2020-07-18:
Thank you for correcting the problem so fast. You have my graditude.
Glenn

closed 2020-07-18 by huyangping