gmag11/EnigmaIOT #8: Node data send error: Registration timeout

Martin09, 2020-07-20:
I've been struggling with a bit of an enigma of my own. I can't seem to get a message sent successfully after I initialize the node. The gateway joins my wifi network okay:

> *WM: Adding parameter
*WM: netkey
*WM: Adding parameter
*WM: channel
*WM: Adding parameter
*WM: netname
*WM: Adding parameter
*WM:
*WM: Adding parameter
*WM: mqttserver
*WM: Adding parameter
*WM: mqttport
*WM: Adding parameter
*WM: mqttuser
*WM: Adding parameter
*WM: mqttpass
*WM:
*WM: AutoConnect Try No.:
*WM: 0
*WM: Connecting as wifi client...
*WM: Try to connect with saved credentials
*WM: Connection result:
*WM: 3
*WM: IP Address:
*WM: 192.168.178.35

My node also seems to initialize okay:
>*WM: WiFi save
*WM: Parameter
*WM: sleepy
*WM: 10
*WM: Parameter
*WM: nodename
*WM: test_node
*WM: Sent wifi save page
*WM: Handle not found
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Waiting for connection result with time out
*WM: Connection result: 
*WM: 3

But then it starts looping and none of the messages get through, as far as I can tell:
>RTC MEM DATA:
 -- CRC: 32 BD E8 49 
 -- Node Key: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 -- Node key is NOT valid
 -- Node status is 0: NOT REGISTERED
 -- Node name: test_node
 -- Last message counter: 0
 -- NodeID: 0
 -- Channel: 6
 -- RSSI: -45
 -- Network name: IreMarIOT
 -- Gateway: 3C:71:BF:5A:A9:29
 -- Comm errors: 0
 -- Network Key: 66 68 7A AD F8 62 BD 77 6C 8F C1 8B 8E 9F 8E 20 08 97 14 85 6E E2 33 B3 90 2A 59 1D 0D 5F 29 25 
 -- Mode: sleepy

>ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
[... 93 lines cut ...]
D [9272][NodeList.cpp:114] reset() Heap: 33944 | Reset packet rate
D [9278][EnigmaIOTNode.cpp:292] saveRTCData() Heap: 33944 | Write configuration data to RTC memory
V [9286][EnigmaIOTNode.cpp:294] saveRTCData() Heap: 33944 | Write RTCData: E5 10 04 C3 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 3C 71 BF 5A A9 29 D4 55 41 A0 B3 44 25 F6 38 34 DD BC C9 42 0E 31 17 1F FA 39 3B E7 17 A7 14 22 35 8C 8B CB 6C D7 A3 49 72 65 4D 61 72 49 4F 54 00 89 28 74 68 F1 BF 2A F3 2C AB 29 01 0A 00 00 00 74 65 73 74 6E 6F 64 65 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 3A 00 00 00 00 00 00 00 00 34 4E 
RTC MEM DATA:
 -- CRC: E5 10 04 C3 
 -- Node Key: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 -- Node key is NOT valid
 -- Node status is 0: NOT REGISTERED
 -- Node name: testnode
 -- Last message counter: 0
 -- NodeID: 0
 -- Channel: 6
 -- RSSI: -44
 -- Network name: IreMarIOT
 -- Gateway: 3C:71:BF:5A:A9:29
 -- Comm errors: 0
 -- Network Key: 55 41 A0 B3 44 25 F6 38 34 DD BC C9 42 0E 31 17 1F FA 39 3B E7 17 A7 14 22 35 8C 8B CB 6C D7 A3 
 -- Mode: sleepy
I [9376][EnigmaIOTNode.cpp:879] handle() Heap: 33944 | Registration timeout. Go to sleep for 9385 ms

If it helps, here's what I am putting in my configs:
[image]
[image]

Any ideas what I can try from here? I don't have any other esp32 boards to try for the gateway, but I will try that once I get some more. Thanks a lot for all your work on this project, seems really promising and exactly what I am looking for my home IOT network!

gmag11, 2020-07-22:
It is normal that first data send before registration fails. Node will retry after successful node hello and server hello messages.

Registration timeout happens when gateway does not reply client hello with server hello. Maybe it's a network key mismatch? It's difficult since latest code changes. And connection result in node is 3 that means authenticated successfully.

Are you using provided mqtt gateway? May you share your node code?

Compile gateway setting debug to verbose (on ide settings, not EnigmaIOTconfig.h) and check it's makes messages. You will find there if gateway does not receive client hello or it finds an error on it.

Martin09, 2020-07-22:
Yes, I forgot to mention, I am using the examples from the latest release:   EnigmaIOTGatewayMQTT and enigmaiot_node. I will give your suggestions a try and get back to you. Thanks!

Martin09, 2020-07-22:
hmm so I am running a bit low on storage space on my board ("Sketch uses 1309897 bytes (99%) of program storage space. Maximum is 1310720 bytes.") so I couldn't go higher than debug level "Info" but this already tells us something:

> [I][EnigmaIOTGateway.cpp:484] configWiFiManager(): 2704 Heap: 231296 ==== Config Portal result ====
[I][EnigmaIOTGateway.cpp:485] configWiFiManager(): 2706 Heap: 231296 Network Name: IreMarIOT
[I][EnigmaIOTGateway.cpp:486] configWiFiManager(): 2715 Heap: 231296 Network Key: 
[I][EnigmaIOTGateway.cpp:487] configWiFiManager(): 2722 Heap: 231296 Channel: 6
[I][EnigmaIOTGateway.cpp:488] configWiFiManager(): 2729 Heap: 231296 Status: true
[I][EnigmaIOTGateway.cpp:489] configWiFiManager(): 2736 Heap: 231296 Save config: no
[I][GwOutput_mqtt.cpp:168] configManagerExit(): 2744 Heap: 231296 ==== Config Portal MQTTGW result ====
[I][GwOutput_mqtt.cpp:169] configManagerExit(): 2753 Heap: 231296 MQTT server: nodered.martin.notreal.ch
[I][GwOutput_mqtt.cpp:170] configManagerExit(): 2762 Heap: 231296 MQTT port: 1883
[I][GwOutput_mqtt.cpp:171] configManagerExit(): 2769 Heap: 231296 MQTT user: martin
[I][GwOutput_mqtt.cpp:172] configManagerExit(): 2777 Heap: 231296 MQTT password: 
[I][GwOutput_mqtt.cpp:173] configManagerExit(): 2784 Heap: 231296 Status: true
[I][EnigmaIOTGatewayMQTT.ino:419] setup(): 3538 Heap: 230064 STA MAC Address: 30:AE:A4:74:1C:30
[I][EnigmaIOTGatewayMQTT.ino:420] setup(): 3539 Heap: 224940 AP MAC Address: 30:AE:A4:74:1C:31
[I][EnigmaIOTGatewayMQTT.ino:421] setup(): 3544 Heap: 229764 BSSID Address: 2C:3A:FD:41:45:95
[I][EnigmaIOTGatewayMQTT.ino:423] setup(): 3552 Heap: 229984 IP address: 192.168.178.34
[I][EnigmaIOTGatewayMQTT.ino:424] setup(): 3560 Heap: 229984 WiFi Channel: 6
[I][EnigmaIOTGatewayMQTT.ino:425] setup(): 3567 Heap: 229984 WiFi SSID: IreMarWifi
[I][EnigmaIOTGatewayMQTT.ino:426] setup(): 3574 Heap: 229984 Network Name: IreMarIOT
[I][GwOutput_mqtt.cpp:213] begin(): 3582 Heap: 229944 CA store set
[I][GwOutput_mqtt.cpp:216] begin(): 3588 Heap: 229944 Set MQTT server nodered.martin.notreal.ch - port 1883
[I][GwOutput_mqtt.cpp:235] reconnect(): 3597 Heap: 229844 Attempting MQTT connection...
[I][GwOutput_mqtt.cpp:445] setClock(): 3607 Heap: 228900 
Waiting for NTP time sync: 
.[I][GwOutput_mqtt.cpp:455] setClock(): 4113 Heap: 229288 Current time: Wed Jul 22 18:27:58 2020

>[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():199]: (-80) UNKNOWN ERROR CODE (0050)
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -80
[E][GwOutput_mqtt.cpp:258] reconnect(): 5271 Heap: 227524 failed, rc=-1 try again in 5 seconds
[E][GwOutput_mqtt.cpp:266] reconnect(): 5278 Heap: 227524 Connect error -80: UNKNOWN ERROR CODE (0050)
[I][GwOutput_mqtt.cpp:235] reconnect(): 10287 Heap: 227556 Attempting MQTT connection...
[I][GwOutput_mqtt.cpp:445] setClock(): 10289 Heap: 227184 
Waiting for NTP time sync: 
[I][GwOutput_mqtt.cpp:455] setClock(): 10292 Heap: 227556 Current time: Wed Jul 22 18:28:04 2020

>[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():199]: (-80) UNKNOWN ERROR CODE (0050)
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -80
[E][GwOutput_mqtt.cpp:258] reconnect(): 10786 Heap: 225968 failed, rc=-1 try again in 5 seconds
[E][GwOutput_mqtt.cpp:266] reconnect(): 10793 Heap: 225968 Connect error -80: UNKNOWN ERROR CODE (0050)
[I][GwOutput_mqtt.cpp:235] reconnect(): 15802 Heap: 226004 Attempting MQTT connection...
[I][GwOutput_mqtt.cpp:445] setClock(): 15804 Heap: 225636 
Waiting for NTP time sync: 
[I][GwOutput_mqtt.cpp:455] setClock(): 15807 Heap: 224296 Current time: Wed Jul 22 18:28:09 2020

...etc.

So it seems like an SSL error? My MQTT server isn't using SSL so I specify port 1883, could that be the issue? Anyways, I will try and see if my MQTT server is working well with another sketch first.

gmag11, 2020-07-23:
You got it. To use MQTT broker without TLS and cetificate you should comment `#include "dstrootca.h"` on `GwOutput_mqtt.h`. [Here](https://github.com/gmag11/EnigmaIOT/blob/master/examples/EnigmaIOTGatewayMQTT/GwOutput_mqtt.h#L21)

gmag11, 2020-07-24:
Gateway code is indeed quite big. But you can fit it loosely by adjusting partition scheme. You should select `MINIMAL SPIFFS` in your IDE, what gives you 1.9MB for code + 1.9MB for OTA and 190kB for filesystem. More than needed for configuration files.

Martin09, 2020-07-26:
Thanks for all the helpful tips! I will try them out in the coming days and report back.

Martin09, 2020-07-27:
Success! (kinda) Indeed now I get it to connect to my MQTT server successfully:

>[I][GwOutput_mqtt.cpp:235] reconnect(): 5919 Heap: 232432 Attempting MQTT connection...
[W][GwOutput_mqtt.cpp:246] reconnect(): 6336 Heap: 229972 MQTT connected
[I][GwOutput_mqtt.cpp:432] publishMQTT(): 6336 Heap: 229972 Publish MQTT. IreMarIOT/gateway/status : 1

I also checked my server and it is receiving the initial message successfully. But node messages are still not getting through. Seems like now I have an encryption error:

>[I][EnigmaIOTGateway.cpp:856] manageMessage(): 108648 Heap: 222432 Reveived message. Origin MAC: 3C:71:BF:5A:A9:28
[I][EnigmaIOTGateway.cpp:874] manageMessage(): 108649 Heap: 222432  <------- CLIENT HELLO
[E][cryptModule.cpp:70] decryptBuffer(): 108656 Heap: 222432 Data authentication error
[E][EnigmaIOTGateway.cpp:1509] processClientHello(): 108663 Heap: 222432 Error during decryption
[E][EnigmaIOTGateway.cpp:899] manageMessage(): 108672 Heap: 222432 Error processing client hello

[If I set the gateway's debug mode to anything higher than "Info" I get core panic crashes. That's why I'm only setting it to "info" here.]

Any ideas what I should check first to start debugging this one? I am happy to open up a separate issue for this error if you think its better. Here is my node output:

>[D][EnigmaIOTNode.cpp:547] begin(): 37 Heap: 293452 Set sleepy mode: sleepy
[V][EnigmaIOTNode.cpp:124] loadRTCData(): 37 Heap: 293452. ----- Read RTCData: 52 EA A7 45 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 30 AE A4 74 1C 31 C3 66 68 7A AD F8 62 BD 77 6C 8F C1 8B 8E 9F 8E 20 08 97 14 85 6E E2 33 B3 90 2A 59 1D 0D 5F 29 25 49 72 65 4D 61 72 49 4F 54 00 00 00 00 00 00 00 00 00 00 00 00 01 0A 00 00 00 74 65 73 74 6E 6F 64 65 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[V][EnigmaIOTNode.cpp:982] checkCRC(): 80 Heap: 293452. CRC32 =  Calc: 0x45A7EA52 Recvd: 0x45A7EA52 Length: 144
[D][EnigmaIOTNode.cpp:149] loadRTCData(): 90 Heap: 293452 Set sleepy mode
 -- Node Key: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
 -- Node key is NOT valid
 -- Node status is 0: NOT REGISTERED
 -- Node name: testnode
 -- Last message counter: 0
 -- NodeID: 0
 -- Channel: 6
 -- RSSI: -61
 -- Network name: IreMarIOT
 -- Gateway: 30:AE:A4:74:1C:31
 -- Comm errors: 0
 -- Network Key: 66 68 7A AD F8 62 BD 77 6C 8F C1 8B 8E 9F 8E 20 08 97 14 85 6E E2 33 B3 90 2A 59 1D 0D 5F 29 25 
 -- Mode: sleepy
[D][EnigmaIOTNode.cpp:552] begin(): 138 Heap: 293452 RTC data loaded. Gateway: 30:AE:A4:74:1C:31
[D][EnigmaIOTNode.cpp:553] begin(): 146 Heap: 293452 Own address: 00:00:00:00:00:00
[D][helperFunctions.cpp:32] initWiFi(): 154 Heap: 293452 initWifi
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][helperFunctions.cpp:51] initWiFi(): 200 Heap: 243144 Mode set to STA. Channel 6
[I][helperFunctions.cpp:58] initWiFi(): 206 Heap: 243144 AP MAC address of this device is 3C:71:BF:5A:A9:29
[I][helperFunctions.cpp:59] initWiFi(): 215 Heap: 243144 STA MAC address of this device is 3C:71:BF:5A:A9:28
[I][espnow_hal.cpp:70] begin(): 225 Heap: 243144 Starting ESP-NOW as node
[D][espnow_hal.cpp:72] begin(): 231 Heap: 243144 Gateway address is 30:ae:a4:74:1c:31
[I][espnow_hal.cpp:40] initComms(): 239 Heap: 243008 Gateway peer Added in channel 6. Result = ESP_OK
[D][espnow_hal.cpp:41] initComms(): 248 Heap: 243008 WIFI channel is 6
[D][EnigmaIOTNode.cpp:630] begin(): 255 Heap: 243008 Comms started. Channel 6
[D][enigmaiot_node.ino:127] setup(): 261 Heap: 243008 Node address set to 3C:71:BF:5A:A9:28
Vcc: 0.000000
Trying to send: 00 02 00 00 01 67 00 CB 02 00 7B 03 73 27 58 04 75 09 7E 
---- Error sending data
Total time: 243 ms
[D][EnigmaIOTNode.cpp:1307] sleep(): 281 Heap: 242744 Sleep programmed for 10000 ms
[D][EnigmaIOTNode.cpp:898] handle(): 1789 Heap: 242780 Current node status: 0
[D][NodeList.cpp:105] reset(): 1789 Heap: 242780 Reset node
[D][NodeList.cpp:114] reset(): 1790 Heap: 242780 Reset packet rate
[I][EnigmaIOTNode.cpp:902] handle(): 1795 Heap: 242780 Random delay (1854)
[V][cryptModule.cpp:143] getDH1(): 5191 Heap: 242796. Public key: 47 7A 81 0F 9C 64 4E 3F 0D 7E 6C 01 E2 D5 5B 7E 4A 50 80 59 6C C6 BA 63 12 88 A5 E8 3E BB BE 2F 
[V][cryptModule.cpp:145] getDH1(): 5195 Heap: 242796. Private key: F0 2B 92 1D 80 8C 75 25 78 D8 C8 54 53 0F 1B 40 A4 FE B1 97 6D 3F 14 18 DD D5 2E 0E 06 A1 72 73 
[V][EnigmaIOTNode.cpp:1028] clientHello(): 5209 Heap: 242796. IV: EC 5D 23 7A 80 D2 95 73 A6 88 16 94 
[D][EnigmaIOTNode.cpp:1039] clientHello(): 5218 Heap: 242796 Signal sleepy node
[V][EnigmaIOTNode.cpp:1047] clientHello(): 5225 Heap: 242796. Client Hello message: FF EC 5D 23 7A 80 D2 95 73 A6 88 16 94 47 7A 81 0F 9C 64 4E 3F 0D 7E 6C 01 E2 D5 5B 7E 4A 50 80 59 6C C6 BA 63 12 88 A5 E8 3E BB BE 2F B1 E2 75 90 
[V][cryptModule.cpp:92] encryptBuffer(): 5245 Heap: 242796. IV: EC 5D 23 7A 80 D2 95 73 A6 88 16 94 
[V][cryptModule.cpp:93] encryptBuffer(): 5254 Heap: 242796. Key: 66 68 7A AD F8 62 BD 77 6C 8F C1 8B 8E 9F 8E 20 08 97 14 85 6E E2 33 B3 
[V][cryptModule.cpp:94] encryptBuffer(): 5266 Heap: 242796. AAD: FF EC 5D 23 7A 80 D2 95 73 A6 88 16 94 90 2A 59 1D 0D 5F 29 25 
[V][cryptModule.cpp:103] encryptBuffer(): 5278 Heap: 242796. Tag: 69 BF 4F B1 9B 65 12 63 11 4B 92 06 C0 E9 DD 6D 
[... cut ...]
[V][EnigmaIOTNode.cpp:303] saveRTCData(): 7342 Heap: 242796. ----- Write RTCData: 52 EA A7 45 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 30 AE A4 74 1C 31 C3 66 68 7A AD F8 62 BD 77 6C 8F C1 8B 8E 9F 8E 20 08 97 14 85 6E E2 33 B3 90 2A 59 1D 0D 5F 29 25 49 72 65 4D 61 72 49 4F 54 00 00 00 00 00 00 00 00 00 00 00 00 01 0A 00 00 00 74 65 73 74 6E 6F 64 65 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[I][EnigmaIOTNode.cpp:879] handle(): 7388 Heap: 242796 Registration timeout. Go to sleep for 6194 ms

gmag11, 2020-08-06:
Data authentication error after HELLO MESSAGE may be caused by wrong network key. Latest version makes key selection simpler and less prone to mistakes. Test latest version, just released 0.9.4

gmag11, 2020-09-10:
As it's been one month after last activity I close this. Feel free to reopen it again if needed

closed 2020-09-10 by gmag11