gmag11/EnigmaIOT #9: Compilation error in helperFunctions.h for ESP8266

cacharritos, 2020-09-06:
Hi @gmag11 , I'm trying to use your EnigmaIoT solution to set up a mesh of battery powered sensors in my garden, and I think yor solution is great for that :-)

I'm getting a compilation error in EnigmaIOTGatewayMQTT.ino configured for an ESP8266:

helperFunctions.cpp:139:23: error: expected initializer before 'extractFileName'

 const char* IRAM_ATTR extractFileName (const char* path) {

                       ^

exit status 1

Arduino:1.8.7 (Windows 10), Tarjeta:"LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

ATENCIÓN: Categoría '' en librería Crypto no es válida. Configurando a 'Uncategorized'
D:\Google Drive\Domotica\ARDUINO\sketches\_ROB_Skethces_MySensors_modificados\libraries\src\helperFunctions.cpp:139:23: error: expected initializer before 'extractFileName'

 const char* IRAM_ATTR extractFileName (const char* path) {

                       ^

exit status 1

Could you please help me to solve it?

Thank you for your support!

gmag11, 2020-09-06:
Error happens in a very simple function. I've never had an issue there. Maybe you have an outdated esp8266 Arduino core. What platform are you trying to use this code if in? What version do you have?

cacharritos, 2020-09-06:
Yes! You are right, that was the problem, I had an outdated version of the ESP8266 board in Arduino IDE. Now it compiles OK!

Thanks a lot for your quick response and tip! 👍

cacharritos, 2020-09-06:
Issue may be closed

closed 2020-09-06 by cacharritos