gmag11/EnigmaIOT #10: Data not received

a-pinon, 2020-09-11:
Hi, 
I have been testing the 0.5.1 with a enigmaiot_node_nonsleepy on ESP8622 with a temperature sensor and two ESP8622 for the Gatway and works ok.
Now, I tried test the 0.9.4 version with the enigmaiot_node_nonsleepy.ino without modification on a ESP8622 and the EnigmaIOTGatewayMQTT.ino on a ESP32 without secure MQTT. 
In the Mosquito MQTT broker i suscribe all the mensages from Gateway, but dont receive the data from node. 
The network name of Gateway is Casa01, and I receive "status" from Gateway, and when I connect the node, receive the hello message, and after that de rssi result from node.
After that, when the node send a data, the  MQTT broker only receive the topic "status", but not the topic "data" from Gateway
This is the data received in the MQTT broker:

11/9/2020 20:30:51node: a57e08c9.0914a8
Casa01/gateway/status : msg.payload : string[1]
"1"
11/9/2020 20:32:32node: a57e08c9.0914a8
Casa01/5C:CF:7F:B5:60:FA/hello : msg.payload : buffer[32]
[ 123, 34, 97, 100, 100, 114, 101, 115, 115, 34 … ]
11/9/2020 20:32:32node: a57e08c9.0914a8
Casa01/W01/hello : msg.payload : buffer[32]
[ 123, 34, 97, 100, 100, 114, 101, 115, 115, 34 … ]
11/9/2020 20:32:34node: a57e08c9.0914a8
Casa01/W01/result/rssi : msg.payload : string[24]
"{"rssi":-44,"channel":1}"
11/9/2020 20:37:22node: a57e08c9.0914a8
Casa01/W01/status : msg.payload : string[75]
"{"per":0.000000e+00,"lostmessages":0,"totalmessages":1,"packetshour":39.70}"
11/9/2020 20:39:02node: a57e08c9.0914a8
Casa01/W01/status : msg.payload : string[75]
"{"per":0.000000e+00,"lostmessages":0,"totalmessages":2,"packetshour":37.25}"
11/9/2020 20:40:42node: a57e08c9.0914a8
Casa01/W01/status : msg.payload : string[75]
"{"per":0.000000e+00,"lostmessages":0,"totalmessages":3,"packetshour":36.55}"
11/9/2020 20:42:22node: a57e08c9.0914a8
Casa01/W01/status : msg.payload : string[75]
"{"per":0.000000e+00,"lostmessages":0,"totalmessages":4,"packetshour":36.27}"
11/9/2020 20:44:02node: a57e08c9.0914a8
Casa01/W01/status : msg.payload : string[75]
"{"per":0.000000e+00,"lostmessages":0,"totalmessages":5,"packetshour":36.14}"
11/9/2020 20:45:42node: a57e08c9.0914a8

The debug information from Gateway when the node is connected, is in the debug file attached named logGateway01. [LogGateway_01.txt](https://github.com/gmag11/EnigmaIOT/files/5211151/LogGateway_01.txt)
The debug output from Gateway goes into a loop only interrumpied when the node send a data. See logGateway 02
[LogGateway_02.txt](https://github.com/gmag11/EnigmaIOT/files/5211154/LogGateway_02.txt)
I compiled the Gateway with "Info" debug level because if I use a sup level the ESP crash.

If I use the enigmaiot_node_msgpack example, the data send by the node are received by the MQTT broker.

Any ideas what I can try from here? 
Thanks a lot for all your work on this project.

gmag11, 2020-09-13:
Logs show normal behabiour, no errors and no warnings. So, I understand communication between node and gateway is fine.

What MQTT client are you using to subscribe to messages? What is the configuration / comman line ?

a-pinon, 2020-09-16:
I am using Mosquitto MQTT.
I have two Gateway-MQTT bridges. One with 0.5.1 version named Home01 and another one with 0.9.4 version named Casa01.
I have one node connected to each of them.
The command line to subscribe is:
> mosquitto_sub -v -h 192.168.0.150 -p 1883 -t '#' 

and the output is:
>Casa01/W01/status {"per":0.000000e+00,"lostmessages":0,"totalmessages":26,"packetshour":362.90}
Home01/5C:CF:7F:23:D7:DC/data [{"channel":0,"type":103,"name":"temperature","value":22.7},{"channel":1,"type":104,"name":"humidity","value":70.5},{"channel":2,"type":121,"name":"altitude","value":20},{"channel":3,"type":115,"name":"pressure","value":1010.8}]
Home01/5C:CF:7F:23:D7:DC/status {"per":0.000000e+00,"lostmessages":0,"totalmessages":2,"packetshour":1060.38}
Casa01/W01/status {"per":0.000000e+00,"lostmessages":0,"totalmessages":27,"packetshour":362.90}
Home01/5C:CF:7F:23:D7:DC/data [{"channel":0,"type":103,"name":"temperature","value":22.7},{"channel":1,"type":104,"name":"humidity","value":70.5},{"channel":2,"type":121,"name":"altitude","value":20},{"channel":3,"type":115,"name":"pressure","value":1010.8}]
Home01/5C:CF:7F:23:D7:DC/status {"per":0.000000e+00,"lostmessages":0,"totalmessages":3,"packetshour":360.36}

In the 0.5.1 version system (Home01) I receive "status" and "data" topics for the node, but in the  0.9.4 version system only receive "status" topic but not "data".

Thanks in advance.

gmag11, 2020-10-11:
Nodes must have same version as gateway they are connected to. You may try to use another example as enigmaiot_node_msgpack.

Sorry, I cannot give you more help without more information.

a-pinon, 2020-10-21:
Thanks for your time.
I used the same version on the node and gateway for testing each version.
The examples are used as is for testing version 0.9.4.
If I use the enigmaiot_node_msgpack example then I get the "data" topic.

gmag11, 2020-10-25:
I'm glad it is working. I'm closing this issue. Let me know if I can help you again

closed 2020-10-25 by gmag11