gmag11/EnigmaIOT #14: DEBUG setting in EnigmaIOTConfig.h has no effect on ESP32 platform

gmag11, 2020-10-11:
@quangvankts asked in #11

DEBUG seems only work on my ESP8266 node but not ESP32 gateway. I tried "#define DEBUG_LEVEL VERBOSE" and "#define DEBUG_ESP_PORT Serial" at the main sketch with no effect. Serial.printf() still works as normal. (Im using Platform IO latest core)

gmag11, 2020-10-11:
That's right, ESP32 platform uses different mechanism to configure debug level.

If you use Arduino you need to configure debug level in board configuration (`core debug level`).

If you use Platformio you may read this section in Platformio docs: https://docs.platformio.org/en/latest/platforms/espressif32.html#debug-level

closed 2020-10-11 by gmag11

quangvankts, 2020-10-16:
Yes I understand how debug configuration work on EPS32 now 
Thanks !