gmag11/EnigmaIOT #15: OTA not working on version 0.9.5

gmag11, 2020-11-06:
OTA is not working on any version. It's not been any change on OTA management since some time ago so it must be something external.

Activity LED does not flash so it seems that it is due to an external library.

gmag11, 2020-11-06:
On 20th may @knolleary published [version 2.8](https://github.com/knolleary/pubsubclient/releases/tag/v2.8) of PubSubClient library. It uses a buffer of 256 bytes to receive mqtt messages.
EnigmaIOTGatewayMQTT solved this by redefining MQTT_MAX_PACKET_SIZE to 2048. After PubSubClient v2.8 is released this solution is not valid anymore. As you can read on PubSubClient release notes, using setBufferSize() this can be fixed.

This will be integrated on version [0.9.6](https://github.com/gmag11/EnigmaIOT/releases/tag/0.9.6). Meanwhile please use PubSubClient version prior to v2.8

knolleary, 2020-11-06:
Hi @gmag11 

you should still be able to use `MQTT_MAX_PACKET_SIZE` to set the packet size. The `setBufferSize()` was added as a way to allow the sketch to change the size from whatever `MQTT_MAX_PACKET_SIZE` was set to.

gmag11, 2020-11-07:
Thank you @knolleary. I'll investigate that.