gmag11/EnigmaIOT #18: ESP32 Gateway crashing in addInputMsgQueue function

arihantdaga, 2021-01-27:
Hello, 
Thank you very much @gmag11  for this project. I started using it recently for one of my application. So far it has been working great, i was using esp8266 gateway. But then due to heap shortage, i decided to switch to esp32 for the gateway. 
But i think something is crashing esp32 in the addInputMsqQueue function. I am realtively new to esp32, so i could not pinpoint the reason. 
I am using a slightly modified version of EnigmaIOTGatewayDummy example. Its working with esp8266 perfectly. 
In esp8266i get this in logs 
```
D [8966][EnigmaIOTGateway.h:157] push() Heap: 24952 | Add element. Buffer was not full

D [8967][EnigmaIOTGateway.h:158] push() Heap: 24952 | Before -- > ReadIdx: 0. WriteIdx: 0. Size: 0

D [8972][EnigmaIOTGateway.h:173] push() Heap: 24952 | After -- > ReadIdx: 0. WriteIdx: 1. Size: 1

D [8981][EnigmaIOTGateway.cpp:815] addInputMsgQueue() Heap: 24952 | Message 0xFF added from 40:F5:20:3E:58:1D. Size: 1
D [8991][EnigmaIOTGateway.h:201] front() Heap: 24952 | Read element. ReadIdx: 0. WriteIdx: 1. Size: 1

D [9000][EnigmaIOTGateway.cpp:834] getInputMsgQueue() Heap: 24952 | EnigmaIOT message got from queue. Size: 1

```

But in esp32

```
[D][EnigmaIOTGateway.h:157] push(): 26627 Heap: 223640 Add element. Buffer was not full

[D][EnigmaIOTGateway.h:158] push(): 26628 Heap: 223640 Before -- > ReadIdx: 0. WriteIdx: 0. Size: 0

[D][EnigmaIOTGateway.h:173] push(): 26634 Heap: 223640 After -- > ReadIdx: 0. WriteIdx: 1. Size: 1

[D][EnigGuru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4008b56e  PS      : 0x00060134  A0      : 0x8008a6ef  A1      : 0x3ffb5810  
A2      : 0x3ffb81f0  A3      : 0x3ffb5d6c  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060123  A7      : 0x00000000  A8      : 0x3ffb5d6c  A9      : 0x3ffb5d6c  
A10     : 0x00000002  A11     : 0x00000002  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060123  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace: 0x4008b56e:0x3ffb5810 0x4008a6ec:0x3ffb5830 0x40088987:0x3ffb5850 0x400e3b78:0x3ffb5890 0x400d8767:0x3ffb58f0 0x400d8795:0x3ffb5a40 0x400e11f5:0x3ffb5a60 0x4013dee9:0x3ffb5a80 0x4010abcf:0x3ffb5aa0 0x4010aab0:0x3ffb5ad0 0x400f5d29:0x3ffb5b00 0x400f602b:0x3ffb5c00 0x4016dc21:0x3ffb5c20 0x4008f616:0x3ffb5c70 0x40088b81:0x3ffb5cb0

```
It did not reach EnigmaIOTGateway.cpp:815. And i saw something like `portENTER_CRITICAL (&myMutex);` in addInputMsgQueue. So i understand something is special here. But i could not understand why is it crashing here.

gmag11, 2021-01-27:
Hi. When in debug mode there is a lot of info coming out from serial port. It may happen that Serial is not fast enough to deliver all data and a bigger overflow occurs, resulting in a watchdog exception and consequent reboot.

Try to reduce debug to warning level and test again. If the error keeps happening you may try to change DEBUG_DBG lines to DEBUG_WARN so you are able to see some messages in warning only mode.

Anyway you can use mqtt gateway code as it is without any change. I've been using it for months with 10 different nodes now. I recommend to use Platformio to compile the code.

arihantdaga, 2021-01-27:
@gmag11 . Thank you. That helped. After lowering debug level, it is not crashing. But i am really curious about it, how could it lead to crash, it's not a memory overflow problem, also, serial in esp32 is definitely faster than esp8266. Is enigmaIoT functions running in timers or some async manner? I was assuming everything is running in the same thread and is handled by EnigmaIOTGateway.handle() method. 
And also why do we need to disable interrupts when pushing in the queue? I am sorry for being so curious. But i just wish to understand it better to contribute further in the development of this library.

gmag11, 2021-01-27:
You can try to increase baudrate to 921600 bps

arihantdaga, 2021-01-31:
Got it working. Thanks. Closing.

closed 2021-01-31 by arihantdaga