gmag11/EnigmaIOT #20: API doesn't send correct json

G2G2G2G, 2021-01-30:
All of the web API sends single quotes instead of double quotes

https://github.com/gmag11/EnigmaIOT/blob/master/docs/api.md

gmag11, 2021-02-02:
Yes, this is intended to simplify string management in C++ code. single quotes are valid in JSON, aren't they? At least Insomnia shows correct decoded JSON data.

G2G2G2G, 2021-02-03:
Yes I saw that in your code, unfortunately no JSON spec is double quotes only, some references here: https://stackoverflow.com/questions/14355655/jquery-parsejson-single-quote-vs-double-quote

I can edit it, if you don't have time, but I would just switch all of the `'` to `\"` instead

gmag11, 2021-02-04:
Thank you. I'll change this in next release.

gmag11, 2021-02-04:
It should be solved in a5b2a76. Thanks again

closed 2021-02-04 by gmag11