gmag11/EnigmaIOT #21: How to send sane messages?

G2G2G2G, 2021-01-31:
I don't see it in any of the examples, best one seems to be:
https://github.com/gmag11/EnigmaIOT/blob/master/examples/enigmaiot_node/enigmaiot_node.ino
which has no control of what is sent?

Instead of:
```
g/1/data [{"channel":0,"type":2,"name":"analog_in","value":2.99},{"channel":1,"type":103,"name":"temperature","value":21.1}]
g/1/status {"per":0.000000e+00,"lostmessages":0,"totalmessages":292,"packetshour":122.46}
```
should be:
```
g/1/data {"analog_in":2.99,"temperature":21.1}
```
JSON is like:
"THING":"VALUE" 
no reason to do
"name":"THING","value":VALUE

What's all the channel, types, etc I can't find documentation at all and I can't think of a use case for it? MQTT has channels (topics) if it's needed to spread them out. And status calculations/storage just seems to waste gateway memory. 

Is this padding for the encryption so it's harder to break or something?

dgcasana, 2021-01-31:
if you are going to work with json objects I recommend you
**enigmaiot_node_msgpack.ino**
that example sends messages as you expect.
you can even send values ​​in array (for DHT22 for example).

```
float humidity = dht.getHumidity();
  float temperature = dht.getTemperature();

  String sensor = String(temperature) + ";" + String (humidity) + ";0"; // dummy data for domoticz

  json["idx"] = 5;
  json["nvalue"] = 0;
  json["svalue"] = sensor;

  int len = measureMsgPack (json) + 1;
	uint8_t* buffer = (uint8_t*)malloc (len);
	len = serializeMsgPack (json, (char*)buffer, len);
	Serial.printf ("Message Len %d\n", len);
	Serial.printf ("Trying to send: %s\n", printHexBuffer (buffer, len));
	// Send buffer data
	if (!EnigmaIOTNode.sendData (buffer, len, MSG_PACK)) {
		Serial.println ("---- Error sending data");
	} else {
		Serial.println ("---- Data DHT22 sent");
	}
	free (buffer);
```

The channel, type, etc is how CayenneLLP treats the messages, before the msgpack I used the channel as idx in domoticz.

About status messages I'm not sure but I think they can be disabled in
**espnow_hall.h**

G2G2G2G, 2021-01-31:
@dgcasana Yes I mentioned that in my post and asked about Cayenne encoding there as well.

Also DHT22 doesn't measure humidity correctly, can be off 3-20%, you should use SHT31, SHT35, SHT85 or BME280 (if you want pressure) 
DHT and AHT sensors aren't usable for anything other than temperature. Do not buy them.

gmag11, 2021-02-02:
CayenneLPP to JSON conversion is done by [CayenneLPP library](https://github.com/ElectronicCats/CayenneLPP) so I've attached to its format. You can use that and process it after receiving MQTT message.

EnigmaIOTNode example shows how to manage CayenneLPP data.

If you need specific JSON format, then you have to use MsgPack.

I do not remember if status data can be disabled. Anyway you can subscribe to g/1/data instead g/1/# to get only data messages. I'll check if there is any setting in EnitmaIOTConfig.h to control that feature. I will add it if does not exist.

G2G2G2G, 2021-02-03:
Actually looks like the channel stuff is some specific design of cayenneLPP, some reason the extra messages are added (idc enough to check why) but it's just extra useless data so I guess I'll have to use msgpack.

closed 2021-02-03 by G2G2G2G

gmag11, 2021-02-04:
In a3e0439ab26b9f2f76b409e871881a97881c1483 i've added an option to EnigmaIOTConfig.h to control status sending after data [here](https://github.com/gmag11/EnigmaIOT/blob/dev/src/EnigmaIoTconfig.h#L28)