gmag11/EnigmaIOT #23: EnigmaIOTGatewayMQTT example not working

nottledim, 2021-03-15:
I'm having trouble getting this example program to work. 

1)  I can't get back to the config page.  The webserver has a presence on port 80 once it's connected to my network but any request url is just "not found".  The only way to reconfigure is to erase flash.  That's a bit drastic!

2) I see `Fouter/gateway/status 1` mqtt message being issued every 5 mins but I don't know what it means.  There is no response to any messages I send such as `Fouter/gateway/get/version`.  If I try to update firmware using python tool it just sits there reporting `Waiting for non sleepy confirmation`.  

3) The config does not allow to connect without mqtt user/pwd.  It should be permitted. 

It's always nice when an example works because it provides a platform to get to know the application.  Can you give me any advise to help get this example working?  

Thanks

dgcasana, 2021-03-16:
1.Your gateway is working fine.
Now you have to connect Nodes to your new esp-now (i think you called "Fouter")
You can see that it is working in your web browser through REST API , just `http://**gatewayIP**/api/gw/info`
[](https://github.com/gmag11/EnigmaIOT/blob/master/docs/api.md)
But normal running is you receive info of your nodes through the mqtt of gateway.
2. `status 1` means gateway is online.
3. You can connect without pwd, only with user  using [](https://github.com/gmag11/EnigmaIOT/releases/download/0.9.7/EnigmaIOT-Gateway-ESP32-Plain_MQTT.bin)

nottledim, 2021-03-16:
Thanks.  I'm still struggling though.  I can't get nodes to connect. The problem seems to be that they're using the mac address of the house network access point as the gateway mac.  If I use the gateway AP it doesn't work because that's on the wrong channel and it has the wrong mac address (since it's a different interface).  I'm also puzzled because the node is connecting to (obtaining IP address from) network AP even though it doesn't need to, indeed shouldn't.

What am I doing wrong?  On the config screen of the node what is it expecting to be entered?

mgeers, 2021-03-16:
I'm having the same issue as mentioned above. I have setup my node on an ESP8266, and am trying to communicate with an ESP32 (gateway). The gateway mac address that the node is trying to connect to is not the MAC address of the ESP32.
Is it worth trying to use 2 ESP32s instead? I have read elsewhere that setting a different channel than channel 1 for the ESP8266 is a bit hit and miss. 
No nodes are registered as per the API call /api/gw/nodes.

dgcasana, 2021-03-17:
As I can understand you have set the same name in both networks (Wi-Fi & EspNow), that's your mistake.
EnigmaIot Gateway creates a new ~Wi-Fi~ network, so you have to reset your Gateway and give a different name to your EnigmaIot network (I called mine "Esp-Now"), and maybe a different password. After that, you can connect the nodes to your new Esp-Now network

nottledim, 2021-03-17:
Of course, lightbulb moment.

nottledim, 2021-03-17:
The lightbulb's gone out.  I can't get it to work it just reports error sending data.

This is the node starting.  Everything seems OK, it's found SSID==fouter which is the name of esp-now network.  It's chosen channel 9, the channel of my home network but my wifi analyzer shows fouter is using channel 6 (ch3 was entered in the config!).
```
10:03:55.947 -> [I][EnigmaIOTNode.cpp:733] begin(): 28 Heap: 309708 Starting from Flash
10:03:56.044 -> [I][EnigmaIOTNode.cpp:745] begin(): 127 Heap: 305364 SPIFFS mounted
10:04:00.468 -> [I][EnigmaIOTNode.cpp:882] searchForGateway(): 4551 Heap: 254188 Gateway fouter found: 1
10:04:00.468 -> [I][EnigmaIOTNode.cpp:883] searchForGateway(): 4551 Heap: 254188 BSSID: 30:AE:A4:07:6C:31
10:04:00.468 -> [I][EnigmaIOTNode.cpp:884] searchForGateway(): 4556 Heap: 254188 Channel: 9
10:04:00.468 -> [I][EnigmaIOTNode.cpp:885] searchForGateway(): 4562 Heap: 254188 RSSI: -57
10:04:00.468 -> [I][helperFunctions.cpp:59] initWiFi(): 4570 Heap: 254500 AP MAC address of this device is 30:AE:A4:1C:B8:BD
10:04:00.502 -> [I][helperFunctions.cpp:60] initWiFi(): 4578 Heap: 254500 STA MAC address of this device is 30:AE:A4:1C:B8:BC
10:04:00.502 -> [I][espnow_hal.cpp:70] begin(): 4588 Heap: 254500 Starting ESP-NOW as node
10:04:00.502 -> [I][espnow_hal.cpp:40] initComms(): 4595 Heap: 254364 Gateway peer Added in channel 9. Result = ESP_OK
10:04:00.502 -> Vcc: 0.000000
10:04:00.535 -> Trying to send: 00 02 00 00 01 67 00 CB 02 00 7B 03 73 27 58 04 75 09 7E 
10:04:00.535 -> ---- Error sending data
10:04:00.535 -> Total time: 4586 ms
10:04:02.030 -> [I][EnigmaIOTNode.cpp:1088] handle(): 6116 Heap: 254000 Random delay (2850)
10:04:06.419 -> [I][EnigmaIOTNode.cpp:1273] clientHello(): 10503 Heap: 254000  -------> CLIENT HELLO
10:04:08.648 -> [I][EnigmaIOTNode.cpp:1055] handle(): 12744 Heap: 254000 Current node status: 2
10:04:08.648 -> [I][EnigmaIOTNode.cpp:1065] handle(): 12745 Heap: 254000 Registration timeout. Go to sleep for 8339 ms
10:04:16.958 -> ets Jun  8 2016 00:22:57
```
I've no idea.

mgeers, 2021-03-17:
This helped me. It's working now. Thanks for your help.

dgcasana, 2021-03-18:
try to reset node _erase_flash_ and reconnect to Gateway.

gmag11, 2021-07-15:
@nottledim Did you get it working. I've written a howto guide that may help to start with EnigmaIOT. Check it here: https://github.com/gmag11/EnigmaIOT/blob/master/docs/howto.md

nottledim, 2021-07-16:
No, I gave up in frustration; resolved my requirements a different way. Now you've reminded me I might try again.  Thanks.

closed 2021-07-16 by nottledim