gmag11/EnigmaIOT #24: Problem with time synchronization in node_nonsleepy

dgcasana, 2021-03-16:
In release 0.9.7 when you synchronize the time you always get "19/01/1970 17:51:34", it always does not change with the passage of time.
The gateway synchronizes well with NTP, its date and time are correct.
here is a screenshot of the node's debug_info. 
```
No sync localtime: 986986500
Timestamp sec: 1615894
19/01/1970 17:51:34
Vcc: 3.050000
Trying to send: 00 02 01 31 01 67 00 CB 
I [3800382][EnigmaIOTNode.cpp:1686] dataMessage() Heap: 38208 | Data message #385
I [3800387][EnigmaIOTNode.cpp:1729] dataMessage() Heap: 38208 |  -------> DATA
---- Data sent
I [3805378][EnigmaIOTNode.cpp:1309] clockRequest() Heap: 38232 | Control message #136
I [3805378][EnigmaIOTNode.cpp:1338] clockRequest() Heap: 38232 |  -------> CLOCK REQUEST
I [3805387][EnigmaIOTNode.cpp:2522] manageMessage() Heap: 38232 | Reveived message. Origin MAC: 3C:61:05:17:18:4D
I [3805392][EnigmaIOTNode.cpp:2652] manageMessage() Heap: 38232 |  <------- CLOCK RESPONSE
I [3805400][EnigmaIOTNode.cpp:1394] processClockResponse() Heap: 38232 | Downlink msg #136
I [3805408][EnigmaIOTNode.cpp:1397] processClockResponse() Heap: 38232 | Accepted
I [3805415][EnigmaIOTNode.cpp:1431] processClockResponse() Heap: 38232 | Offest adjusted to 970 us, Roundtrip delay is 36620
I [3805426][EnigmaIOTNode.cpp:2655] manageMessage() Heap: 38232 | Clock Response OK
No sync localtime: 986996502
Timestamp sec: 1615894
19/01/1970 17:51:34
Vcc: 3.050000
Trying to send: 00 02 01 31 01 67 00 CB 
I [3810383][EnigmaIOTNode.cpp:1686] dataMessage() Heap: 38208 | Data message #386
I [3810388][EnigmaIOTNode.cpp:1729] dataMessage() Heap: 38208 |  -------> DATA
---- Data sent
```

gmag11, 2021-05-05:
Did you get it working? I need to test on my local setup

GeoffreyMills, 2021-08-08:
Confirmed with ESP8266 as Node

The change in timeManager.h resolves this: 
timeManager.h
time_t unixtime () {
        // uint64_t time_sec = clock () / 1000000L;  // <-- returns 1970, returned result is lacking last 3 digits
		// fix?
		uint64_t time_sec = clock () / 1000L;
        return time_sec;

Possibly more to it than this because quote unixtime()  " returns `millis() / 1000` if not synchronized, local clock otherwise" but I cannot find this logic/code anywhere? 
FYI: Gateway is synced to NTP time (not using certificates or SECURE_MQTT)
API api/node/node for this ESP8266 node shows Timesync: true
Examples Node Sensor code L164 : void setup()  EnigmaIOTNode.enableClockSync (true);         <--was set to false

Geoffrey

gmag11, 2021-08-24:
Thank you @GeoffreyMills. That's right. I'll correct it in next commit

gmag11, 2021-08-24:
Fixed on https://github.com/gmag11/EnigmaIOT/commit/9569a12982a891ec90fdc3c7fa5758b7de8c3151

Tested both on ESP32 and ESP8266

closed 2021-08-24 by gmag11