gmag11/EnigmaIOT #25: esp32-MQTTGateway not compiling

mgeers, 2021-03-17:
I have started with a clean install and tried compiling the esp32-MQTTGateway. I'm getting these errors:

```
Compiling .pio\build\esp32-MQTTGateway-NoSSL\lib70a\src\espnow_hal.cpp.o
src\espnow_hal.cpp: In member function 'virtual void Espnow_halClass::initComms(peerType_t)':
src\espnow_hal.cpp:38:19: error: cannot convert 'esp_interface_t' to 'wifi_interface_t' in assignment
   networkGw.ifidx = ESP_IF_WIFI_STA;
                   ^
src\espnow_hal.cpp:40:19: warning: unused variable 'result' [-Wunused-variable]
         esp_err_t result = esp_now_add_peer (&networkGw);
                   ^
src\espnow_hal.cpp: In member function 'bool Espnow_halClass::addPeer(const uint8_t*)':
src\espnow_hal.cpp:92:13: error: cannot convert 'esp_interface_t' to 'wifi_interface_t' in assignment
  peer.ifidx = ESP_IF_WIFI_AP;
```

I have replaced ESP_IF_WIFI_STA with WIFI_IF_STA, and ESP_IF_WIFI_AP with WIFI_IF_AP and it works. Not sure though if I'm introducing other problems thought that are not immediately obvious to me.

Thanks

dgcasana, 2021-03-18:
This happens in Platformio with the _platform espressif32 @ 3.1.0_.
if you downgrade to v 3.0.0 (you have to uninstall) it compiles without problems, however, gmag11 may have to modify it for future versions.

gmag11, 2021-05-05:
Yes, I try to stay in line with latest released Espressif platform. @mgeers Your solution is right if you want to stay in 3.0.0

closed 2021-05-05 by gmag11