gmag11/EnigmaIOT #27: Wrong use of searchForGateway (&rtcmem_data) ?

arihantdaga, 2021-05-05:
Hi @gmag11 , I came across these if blocks 
`if (searchForGateway (&rtcmem_data), true)` in lines 
https://github.com/gmag11/EnigmaIOT/blob/db1feacc9287ae786eba8caf87142d031d123706/src/EnigmaIOTNode.cpp#L752
https://github.com/gmag11/EnigmaIOT/blob/db1feacc9287ae786eba8caf87142d031d123706/src/EnigmaIOTNode.cpp#L761

was it meant to be - `if (searchForGateway (&rtcmem_data, true))`
I noticed this problem because by node was always trying to conenct to the gateway even when it could not find the gateway. 
It would print this in the logs. 
```
 Gateway SMART_S not found
Found gateway. Storing
```

gmag11, 2021-05-05:
I guess you are right. Would you mind to do a pull request over dev branch? Thank you!

arihantdaga, 2021-05-05:
Sure, yeah.

gmag11, 2021-05-08:
Merged. Will be published in next release

closed 2021-05-08 by gmag11