gmag11/EnigmaIOT #29: Support for BMP280 sensor

ator1811, 2021-06-06:
Hi, impressive work you have done. I try to understand how to make a sensornode with a BMP280 pressure/temperature sensor, but I get lost in the number of examples. So if you can help me a little...?

What example node should I use as a starting point? And where do I put library, setup and loop functions for this sensor? I like to use the EnigmaIOT MsgPack example because I have no experience with Cayenne.

Thx in advance.
Aernout

gmag11, 2021-06-27:
Right now I'm preparing a how-to-start guide to help others to begin programming their own nodes. It will take a while though. I hope less than one month.

Anyway I will leave some guidelines here. Let me know if it is useful and what challenges you find. It will be useful for my guide.

You can start with a regular Arduino Sketch where you design what you want your node does. A simple `setup()` and `loop()` is enough. You can use `Serial.printf()` where you will finally place message sending.

Point of start is [EnigmaIOT JSON Controller Template](https://github.com/gmag11/EnigmaIOT/tree/master/examples/EnigmaIOT-Json-Controller-Template). It is an empty but already configured template where you can add your code.

Inside BasicController.cpp there are a `setup ()` and `loop ()`methods where you can insert your code. You can rename [`CONTROLLER_CLASS_NAME`](https://github.com/gmag11/EnigmaIOT/blob/master/examples/EnigmaIOT-Json-Controller-Template/BasicController.h#L15) if you like, but it is not mandatory.

You can look at [EnigmaIOT-Sensor-Controller example](https://github.com/gmag11/EnigmaIOT/tree/master/examples/EnigmaIOT-Sensor-Controller). It is very close what you need but using DS18B20 instead of BMP280.

gmag11, 2021-07-08:
@ator1811 First version of how to guide is finished in dev branch. Check it there. I will be releasing last version before 1.0 in a few days anyway.

gmag11, 2021-07-15:
How to guide is released in main branch: https://github.com/gmag11/EnigmaIOT/blob/master/docs/howto.md

closed 2021-07-15 by gmag11