gmag11/EnigmaIOT #30: EnigmaIOT beginner's guide or help sources?

scottcopus, 2021-07-04:
EnigmaIOT seems like a great product and something that matches what I've been looking for.  But where does one go to get help?

I've got the gateway binary loaded onto one of my ESP32's (a TinyPico) and I think I've got that running.  But I'm completely lost on how to get a node working.  I think the problem is that I'm a noob with PlatformIO and I can't get any Enigma code to compile.  Do you have a beginner's guide on how to set up PlatformIO and get your sources and dependent libraries into a project and get it to compile?  BTW, my dev environment is Win10.

Are there any precompiled example node binaries that behave the same as the gateway with a similar "setup" process?  It would be great to test out the complete environment without compiling anything and have a node binary to do simple LED on or off or respond to GPIO inputs on various pins and relay via MQTT, etc.

@gmag11 I couldn't find any contact information besides what's listed on GitHub page.  Is that ok? Twitter? Discord?

cheers!
Scott

gmag11, 2021-07-04:
Hello @scottcopus. #29 asked for something similar. I threw a few guidelines there that may help. JSONController examples may illustrate how to develop simple (and not so simple) nodes by using a template as start up.

I'm writting a howto guide. It is partially available on dev branch, under docs directory. It is not complete yet. I'm right now working on it.

Check it out and keep an eye on it on the next few days.

scottcopus, 2021-07-05:
@gmag11 Sounds great!  Having part of the guide discuss setting up the dev environments (Arduino & PlatformIO) from a noob perspective would be nice.  If you've got good links to other sites that can relate to setting up the environments with respect to being able to compile EnigmaIOT, that would be useful too.  For example I don't know if I'm supposed to select "Import Arduino Project" or just "Open Project" from PlatformIO.  I assume the latter since the repo already contains a 'platform.ini'.  But then I'm not sure what I'm really supposed to do next.  Can you give a basic rundown of the steps needed?  Do you have the ability to set up a "fresh" PlatformIO environment to be able to emulate common issues that a noob would experience while trying to download & compile?

Here's my first roadblock when trying to build it:

```
Library Manager: Installing git+https://github.com/gmag11/CryptoArduino.git
FileNotFoundError: [WinError 2] The system cannot find the file specified:
  File "C:\Users\Admin\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Admin\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Admin\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
```
What's your thoughts on my previous suggestion to have a pre-built binary for a node that could send and receive some simple MQTT commands using external input/output?  If you've already got an example that does something like this what about compiling and providing it like you do for the gateway binaries?

FYI, I've been trying out Tasmota for the past couple months.  I wouldn't have been able to use it without their prebuilt binaries though.  However there's a lot of projects where I need something really custom and Tasmota doesn't really fit my needs.  That's when I came across your project.  It sounds very promising!  I just need help getting past over the dev/build hurdles, lol.

thanks!
Scott

gmag11, 2021-07-08:
I guess you don't have GIT utility installed. Same happened to me a few days ago. If you don't get git download working you can download CryptoArduino from [GitHub ](https://github.com/gmag11/CryptoArduino) and copy it under your lib directory in your Platformio project.

Now, how to guide is finished. Feel free to send me any suggestion about missing subjects. Please do it as separate issues.

Personally, I do not see any advantage on having a closed bin of a node. EnigmaIOT has evolved to be a framework to build a cheap network sensor. There are a bunch of examples with different complexity. What I think that is missing are platformio.ini files on every example so that any can compile them out of the box.

gmag11, 2021-07-15:
How to guide is already released in main branch

closed 2021-07-15 by gmag11