gmag11/EnigmaIOT #32: Errors when DEBUG enabled

GeoffreyMills, 2021-07-10:
Hi,
Using Arduino IDE - Compiling the MQTT Gateway with debug enabled (INFO or DEBUG) fails

\EnigmaIOT\src\GatewayAPI.cpp:122:73: error: the address of 'addr' will always evaluate as 'true' [-Werror=address]
      DEBUG_INFO ("Node to process is %s", addr ? p->value ().c_str () : "NULL");

Hack - Commented out L122

EnigmaIOT\src\EnigmaIOTNode.cpp:2247:46: error: 'numBytes' was not declared in this scope
    DEBUG_INFO ("%u bytes written. Total %u", numBytes, totalBytes);
	
#if DEBUG_LEVEL >= INFO
			size_t numBytes = 
#endif

set numBytes = 0; Moved #endif to end of block

compiled OK, running, going to add a node next
WiFi process setup is very nice
Geoffrey Mills

gmag11, 2021-07-15:
Thanks, I'll check it.

gmag11, 2021-08-24:
Your hack is correct.

I've corrected directive condition so that no hack is needed.

Fixed on https://github.com/gmag11/EnigmaIOT/commit/301518b982dc19ac5508e912838eb7a9b806925a

closed 2021-08-24 by gmag11