gmag11/EnigmaIOT #37: Can’t work with EnigmaIOT

etiel22, 2021-10-09:
My setup,
Arduino IDE 1.8.16
Esp8266 board 3.0.2
I’ve installed all the external libraries as you mention on “howto”.
EnigmaIOT version 0.9.8.
“EnigmaIOTGatewayMQTT.ino” compile fine in “WEMOS lolin D1 mini”, (the led flash) but I can´t configure them.
 
The web page is not correct and when I click “Configure WiFi” opens a blank page.

[image]

In debug IDE terminal: 
_*WM: Configuring access point... 
*WM: EnigmaIoTGateway
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: About to scan()
*WM: About to scan()
*WM: Scan done
*WM: About to scan()
*WM: About to scan()_ 

I appreciate your help.
Carlos Leite

steven52880, 2021-11-07:
I have the same problem.
When I connect to the wifi for settings, It shows a wrong page and my esp8266 crashed and reboot.
Besides, the old version(0.9.7) can run correctly.

GeoffreyMills, 2021-11-13:
I have ESP8266 (3.01) and ESP32 working but took me a long time to get this initial WiFi working, though mostly ESP32 issues.. Make sure you are using ESPAsyncWiFiManager -- https://github.com/gmag11/ESPAsyncWiFiManager **forked from** https://github.com/alanswx/ESPAsyncWiFiManager (and I edited this for ESP32)
ESP8266 - use "Generic ESP8266 Module" in Arduino IDE 
Tools ->Erase Flash - erase ALL flash contents PLUS you have to jumper specific pins on ESP8266 as well (suspect this wasted  a lot of my time, I was rebooting back to same code )
Minor: Avoid setting full debug on and setting speed to 115200
Have to say EnigmaIOT is working wonderfully, uptime/reliability is superb
Geoffrey Mills

etiel22, 2021-11-21:
I finally found time to test "enigmaIOT" again, I installed version 0.97 and it works fine, it's a good job by Germán Martín.
Thanks for your support.

Carlos Leite

closed 2021-11-21 by etiel22

steven52880, 2021-11-23:
> I finally found time to test "enigmaIOT" again, I installed version 0.97 and it works fine, it's a good job by Germán Martín. Thanks for your support.
> 
> Carlos Leite

V0.97 does not seem to work with Hass. I bought some esp32 modules, and the gateway sketch works well with them.