gmag11/EnigmaIOT #39: MQTT command line to switch LED off/on

GeoffreyMills, 2021-11-17:
Hi,
FYI This is a documentation change/advisory maybe??
I wanted to control an LED on a node, so I used your code (under examples LED Controller) and added this to the node software I have been using (with a BME280 sensor). I am running Mosquitto (and Thingsboard) on a Centos VM, so I used the linux command line below to switch the LED status:
 mosquitto_pub -h localhost  -t Mesh/6130/set/data -m "{"cmd":"led","led":1}"

The node did not like this, the data was not recognised as MSG_PACK or CAYENNELPP. But other MQTT commands work e.g:
 mosquitto_pub -h localhost  -t Mesh/6130/get/version -m " "

I was reading the ArduinoJSON website about deserializing JSON Objects when I saw the formatting on their JSON examples:
char json[] = "{\"hello\":\"world\"}";

So I formatted the mosquitto_pub command line to match:
mosquitto_pub -h localhost  -t Mesh/6130/set/data -m "{\"cmd\":\"led\",\"led\":1}"

and viola! working now  :)
processRxCommand() | Data: {"cmd":"led","led":1}
[BME280Controller.cpp:176] processRxCommand() | Set LED status to ON

Hmm, seems so obvious now... might save someone some time one day

Regards
Geoffrey Mills

gmag11, 2021-12-25:
Hi @GeoffreyMills ,

You are right. I'll fix it in dev branch, so that it is included on next release. Thank you!!!

gmag11, 2021-12-25:
Fixed on 33e2c7f4f1826dc99ac059278e537e83aae06c24

closed 2021-12-25 by gmag11

GeoffreyMills, 2021-12-26:
Hi,
I read my original notes and I was confused by what I had written. Then I realised that the html formatting ruined another key point.
Inside the { } every speechmark must be prefixed with a backslash, see the formatting displayed here: https://arduinojson.org/v6/api/json/deserializejson/ - scroll down to Examples

So Linux bash shell I am using:
mosquitto_pub -h localhost  -t Mesh/B994/set/data -m "{\\"cmd\":\\"led\",\\"led\\":0}"

With this formatting you get a response (confirmation /data ... ) back via MQTT 
e.g send this command via Linux command line:
mosquitto_pub -h localhost  -t Mesh/B994/set/data -m "{\\"cmd\":\\"led\",\\"led\\":1}"

I ran up another bash shell and used this command line to show the MQTT traffic:   mosquitto_sub -h localhost -v -t '#'    

So running the mosquitto_pub command above returns (in the mosquitto_sub traffic):
Mesh/B994/set/data {"cmd":"led","led":1}      <- we sent this command
Mesh/B994/data {"cmd":"led","led":1}            <- MQTT sends back confirmation

If you omit the backslash then you do not get the confirmation back via MQTT:
Mesh/B994/data {"cmd":"led","led":1}            <- MQTT sends back confirmation

Regards
Geoffrey