gmag11/EnigmaIOT #52: El Gateway MQTT EnigmaIOT no funciona correctamente.

ovallepa, 2023-07-05:
He descargado el proyecto EnigmaIOT y estoy compilando sus ejemplos (Los nodos en ESP8266, El Gateway en ESP32 y el broker lo alterno entre un Odroid C1+ y un tv box con Armbian instalado) hasta ahora los ejemplos compilan sin errores pero no he logrado poner en marcha la red. Aún no he entrado a fondo a solucionar el error pero en el código de: 
* @file EnigmaIOTGatewayMQTT.ino
  * @version 0.9.8
  * @date 15/07/2021
  * @author German Martin
  * @brief MQTT Gateway based on EnigmaIoT over ESP-NOW
  Una vez se carga al ESP32 este entra en punto de acceso como debe ser con WifiManager sin embargo al ingresarle los datos de autenticación no sale del modo punto de acceso y no se conecta a la red, envia el mensaje "Handle not found" y vuelve a mostar el portal cautivo. Favor revisar el ejemplo.

dgcasana, 2023-07-05:
El portal cautivo es la librería wifimanager (creo que se llama así) la
usan muchos proyectos por lo que yo buscaría el error por la parte del
punto de acceso:
- Caracteres raros
- Contraseña demasiado larga
- Mismo SSID para 2.4 y 5 GHz

Prueba creando un punto de acceso con el móvil y una clave sencilla para
ver si conecta.

Pd. a mí con los nodos casi siempre me falla el primer intento pero a la
segunda enlaza.

ovallepa, 2023-07-05:
> El portal cautivo es la librería wifimanager (creo que se llama así) la usan muchos proyectos por lo que yo buscaría el error por la parte del punto de acceso: - Caracteres raros - Contraseña demasiado larga - Mismo SSID para 2.4 y 5 GHz Prueba creando un punto de acceso con el móvil y una clave sencilla para ver si conecta. PD. a mí con los nodos casi siempre me falla el primer intento pero a la segunda enlazar. El mié, 5 jul 2023, 21:34, Efrain Ovalle ***@***.***> escribió:
> […](#)

De hecho cargó algunos ejemplos de la libreria wifimanager y funcionan muy bien con la red wifi existente. Por lo cual creo que el error es intrinseco al ejemplo  EnigmaIOTGatewayMQTT.ino

ovallepa, 2023-07-10:
He seguido realizando pruebas aún no consigo conexión con la red. En un ESP8266, he cargado el ejemplo: 
 * @file enigmaiot_node_nonsleepy.ino
  * @version 0.9.8
  Utilizo el IDE Arduino 2.1.1 y la version de ESPAsync_WiFiManager 1.15. He intendado conectarme a varias redes con SSID de solo caracteres alfanúmericos y aún así no se conecta. En la consola se observa el error:  
12:43:53.713 -> E [102140][H:17096][EnigmaIOTNode.cpp:781] begin() | Configuration error. Restarting
12:43:53.914 -> W [87][H:22648][EnigmaIOTNode.cpp:346] loadFlashData() | /config.json do not exist
Luego del error, el ESP se reinicia y sigue en modo punto de acceso. Son redes 2.4Hz y las contraseñas de 8 caracteres alfanumericos. He creado redes con el movil y no se conecta. Deberian revisar la biblioteca.

ovallepa, 2023-07-11:
Luego de revisar los archivos de la biblioteca EnigmaIOT he logrado que el Gateway se conecte a la red wifi y que el Nodo tambien se autentique, En ambos casos comenté la linea "wifiManager->setTryConnectDuringConfigPortal (false);" que se encuentra en los archivos EnigmaIOTGateway.cpp y EnigmaIOTNode.cpp. Al hacer la revisión me he dado cuenta de gran trabajo que es el proyecto EnigmaIOT y del gran conocimiento que debe tener el autor: German Martin, mi admiración, es un proyecto bastante complejo pero se nota el esfuerzo por simplificarlo. Sin embargo, falta la continuidad en el tiempo y actulizaciones, en mi caso no he logrado que los ejemplos enigmaiot_node_nonsleepy.ino y EnigmaIOTGatewayMQTT.ino funcionen correctamente. En este momento el nodo arranca bien luego que sale de ESPAsyncWiFiManager.h, arma el paquete y solicita la llave para registrarse con el Gateway pero luego  "---- Error sending data" lo cual al aparecer indica que el gateway no le está respondiendo. Por otro lado el Gateway tiene problemas en el código para guardar en el SPIFFS, una vez se ingresan los datos con ESPAsyncWiFiManager.h se reinicia (A veces indefinidamente, a veces correctamente) y muestra:

17:42:13.429 -> [  1049][W][GwOutput_mqtt.cpp:159] loadConfig(): [EnigmaIOT] 1049 Heap: 250264 /mqtt.json do not exist
17:42:13.429 -> [  1049][W][EnigmaIOTGatewayMQTT.ino:421] setup(): [EnigmaIOT] 1049 Heap: 250264 Error reading config file
17:42:13.467 -> [  1056][D][EnigmaIOTGateway.cpp:775] begin(): [EnigmaIOT] 1057 Heap: 249420 Broadcast key: 2C 23 D0 DB 2C DC B4 62 00 CB 6D 0F F4 38 EB 6C 2E 70 FC 97 4F 58 4C A4 C8 6F B9 8A 01 A8 A3 A4 
17:42:13.467 -> [  1073][W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
17:42:13.467 -> [  1086][W][EnigmaIOTGateway.cpp:709] loadFlashData(): [EnigmaIOT] 1086 Heap: 249420 /config.json do not exist
17:42:13.504 -> [  1104][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 0 - WIFI_READY
17:42:13.586 -> [  1198][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
17:42:13.586 -> [  1199][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
17:42:13.586 -> [  1199][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 2 - STA_START
17:42:13.618 -> [  1217][W][EnigmaIOTGateway.cpp:718] loadFlashData(): [EnigmaIOT] 1217 Heap: 203036 Dummy STA config loaded
17:42:13.618 -> *WM: Adding parameter
17:42:13.618 -> *WM: netkey
17:42:13.618 -> *WM: Adding parameter
17:42:13.618 -> *WM: channel
17:42:13.618 -> *WM: Adding parameter
17:42:13.618 -> *WM: netname
17:42:13.618 -> *WM: Adding parameter
17:42:13.618 -> *WM: 
17:42:13.618 -> *WM: Adding parameter
17:42:13.618 -> *WM: mqttserver
17:42:13.618 -> *WM: Adding parameter
17:42:13.618 -> *WM: mqttport
17:42:13.618 -> *WM: Adding parameter
17:42:13.618 -> *WM: mqttuser
17:42:13.618 -> *WM: Adding parameter
17:42:13.659 -> *WM: mqttpass
17:42:13.659 -> *WM: 
17:42:13.659 -> *WM: AutoConnect Try No.:
17:42:13.659 -> *WM: 0
17:42:13.659 -> *WM: Connecting as wifi client...
17:42:13.659 -> *WM: Try to connect with saved credentials
17:42:13.659 -> [  1265][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
17:42:13.659 -> E (2831) wifi:sta is connecting, return error
17:42:13.659 -> [  1273][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007
17:42:16.009 -> [  3624][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: 0, BSSID: 00:00:00:00:00:00, Reason: 201
17:42:16.009 -> [  3624][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
17:42:16.009 -> [  3631][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 201 - NO_AP_FOUND
17:42:16.053 -> [  3638][D][WiFiGeneric.cpp:975] _eventCallback(): WiFi Reconnect Running
17:42:16.053 -> [  3646][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
17:42:16.053 -> *WM: Connection result: 
17:42:16.053 -> *WM: 1
17:42:18.069 -> *WM: AutoConnect Try No.:
17:42:18.069 -> *WM: 1
17:42:18.069 -> *WM: Connecting as wifi client...
17:42:18.069 -> *WM: Try to connect with saved credentials
17:42:18.069 -> [  5680][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
17:42:18.069 -> E (7251) wifi:sta is connecting, return error
17:42:18.069 -> [  5693][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007
17:42:18.100 -> *WM: Connection result: 
17:42:18.100 -> *WM: 1
17:42:18.459 -> [  6067][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: 0, BSSID: 00:00:00:00:00:00, Reason: 201
17:42:18.459 -> [  6067][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
17:42:18.459 -> [  6074][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 201 - NO_AP_FOUND
17:42:18.459 -> [  6081][D][WiFiGeneric.cpp:979] _eventCallback(): WiFi AutoReconnect Running
17:42:18.490 -> [  6089][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
17:42:20.100 -> *WM: AutoConnect Try No.:
17:42:20.100 -> *WM: 2
17:42:20.100 -> *WM: Connecting as wifi client...
17:42:20.100 -> *WM: Try to connect with saved credentials
17:42:20.100 -> [  7710][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
17:42:20.100 -> E (9281) wifi:sta is connecting, return error
[... cut ...]
17:42:20.175 -> [  7769][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 10 - AP_START
17:42:20.645 -> *WM: AP IP address: 
17:42:20.645 -> *WM: 192.168.4.1
17:42:20.645 -> *WM: HTTP server started
17:42:20.645 -> *WM: About to scan()
17:42:20.645 -> *WM: About to scan()

Acá se notan varios errores. N obstante, de tanto hacer pruebas he logrado que se conecte a la red wifi pero una vez se reinicia hay que volver a autenticar. 

Creo que estos problemas vienen por las nuevas versiones de las bibliotecas. En fin, agradeceria su asesoria

ovallepa, 2023-07-12:
Bueno seguí haciendo pruebas y decubrí que hay una función que define si se guardan los datos de autenticación tanto de wifi como de mqtt de forma permanente, la función está en EnigmaIOTGateway.cpp y  es:
void EnigmaIOTGatewayClass::doSave (void) {
	DEBUG_INFO ("Configuration saving activated");
	shouldSave = true;
}
Dicha función se invoca a través de wifiManager->setSaveConfigCallback (doSave); para ser llamada en la libreria de ESPAsyncWiFiManager.cpp mediante la función _savecallback(); cuando debe ser pero aún no encuentro la razón exacta del porque ese llamado de funciones entre librerias no es efectiva y la variable shouldSave no pasa a true. De momento forzadamente hago llamado a doSave(); en  bool EnigmaIOTGatewayClass::configWiFiManager (). De esta forma he logrado la estabilida con el wifi y la comunicación con el broker. No obstante aún el nodo queda en la busqueda del Gateway pero no lo encuentra. Seguiré revisado.

dgcasana, 2023-07-16:
Llevo usando EnigmaIoT desde la versión 0.5 (más de 2 años), y he ido actualizando con cada nueva versión, de modo que desde hace tiempo tengo un fork con los nodo que he programado y últimamente actualizo directamente desde mi fork.

Ante tu insistencia he realizado una prueba desde cero con el Git oficial (éste), desde VisualStudioCode+Platformio he clonado el repositorio de gmag12 he compilado el ejemplo "EnigmaIOTGatewayMQTT" en un Esp32, la ÚNICA modificacion ha sido añadir ".cpp" al ".ino" por un error del compliador (hay otras soluciones pero esta es la mas sencilla) y me ha costado 3 intentos conectar a mi Wi-Fi; los dos primeros intentos :

```
10:43:17.463 > *WM: Sent wifi save page
10:43:19.463 > *WM: Connecting to new AP
10:43:19.463 > *WM: Connecting as wifi client...
10:43:19.464 > *WM: Waiting for connection result with time out
```
A la tercera ha conectado.

Seguidamente he compilado "enigmaiot_node_msgpack" en un esp8266 y me ha costado 4 intentos conectar al Gateway. IMPORTANTE no se si te he entendido mal el mensaje del 10 de julio, pero los nodos se conectan a la red wifi que crea el Gateway;  en este caso me ha costado 4 intentos que el nodo se conectara al Gateway 

```
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result:
*WM: 1
*WM: Failed to connect.

```
A la cuarta la respuesta fue:
```
*WM: Connection result:
*WM: 3
*WM: Connected
```

Por lo que te puedo asegurar que el proyecto funciona correctamente.

Por aportar algo mas de información el esp8266 almacena la información de las credeciales Wifi en la memoria spiff por lo que con un "erase_flash" no se borran, si has utilizado el esp8266 con anterioridad puedo resultar que este intentando conectar con las credenciales anteriormente usadas.
He creado un pequeño programa para resetear mis nodos cuando hago alguna prueba, si no esta en mi GitHub intentaré subirlo y dejarlo publico para que puedas limpiar completamente los esp8266.

PD: para el Gateway mejor usar un Esp32

ovallepa, 2023-07-17:
> Llevo usando EnigmaIoT desde la versión 0.5 (más de 2 años), y he ido actualizando con cada nueva versión, de modo que desde hace tiempo tengo un fork con los nodo que he programado y últimamente [...]

Muchas gracias por responder.  Tal cual usted lo comenta volví a descargar el proyecto y lo compilé en  VisualStudioCode+Platformio y ¡Erureka!, todo funciona. El ejemplo "EnigmaIOTGatewayMQTT" en un Esp32 y "enigmaiot_node_nonsleepy" en un esp8266. Como debe ser solo descargar compilar y todo funcionó bien. Estaba probando todo desde el Ide de Arduinio 2.1.1 y 1.8.13, pero todo se desencadenó en un error tras otro, no pensé que cambiar de entorno hiciera la diferencia. Ya habia conseguido comunicar el nodo con el gateway pero cuando este enviaba al servidor MQTT el esp32 se reiniciaba, o hablaba con el nodo o con el servidor MQTT, de lo contrario terminaba en un reinicio. Al parecer el proyecto tiene dificiltades con el IDE de Arduino, pero con VisualStudioCode+Platformio va excelente.

Estudiando los archivos de la libreria encontré una novedad que se puede mejorar: Sucede que al definir en EnigmaIoTconfig.h con la variable NUM_NODES el número máximo de nodos a usar en el sistema automaticamente el compilador calcula la RAM que se requiere para los mismo  puesto que se definen en uno ciclos en el archivo NodeList.cpp, eso está bien para el Gateway pero en dicho archivo hay un par de funciones que comparte el Gateway con los Nodos haciendo que al cargar ese número máximo de nodos se afecte la RAM del ESP8266 de los Nodos. 

No sé si ya es algo que habias contemplado pero entre las soluciones veo dos una rápida y una eficiente. La rápida seria cambiar el valor de NUM_NODES a 1 o 2 cuando se va a programar un Nodo. Y luego al programar un Gateway colocar el número máximo real estimado. Lo eficiente seria separar ese archivo en dos uno que solo se llame para los nodos y otro que solo se llame para los Gateway. 

De ante mano, Gracias.